syssi/esphome-jbd-bms #1: Setting up UART causing panic?

Bretthansenpc, 2022-05-22:
I don't know if I'm doing something wrong but my setup wouldn't start with the original UART pins so I added the Debug setup and i get this in the logs. I'm not that great at reading this any help would be greatly appreciated! 

[06:00:15][I][app:029]: Running through setup()...
[06:00:15][C][uart.arduino_esp32:077]: Setting up UART...
[06:00:15]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[06:00:15]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[06:00:15]Core 1 register dump:
[06:00:15]PC      : 0x40109bc6  PS      : 0x00060936  A0      : 0x8010e931  A1      : 0x3ffb0060  
[06:00:15]A2      : 0x3ffb0070  A3      : 0x3f401f39  A4      : 0x00000001  A5      : 0x00000018  
[06:00:15]A6      : 0x00000000  A7      : 0x0000000c  A8      : 0x3ffb0070  A9      : 0x3ffb0070  
[06:00:15]A10     : 0x3ffb0070  A11     : 0x0000002a  A12     : 0xffffffff  A13     : 0x3f40aaa0  
[06:00:15]A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000001  
[06:00:15]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[06:00:15]
[06:00:15]ELF file SHA256: 0000000000000000
[06:00:15]
[06:00:15]Backtrace: 0x40109bc6:0x3ffb0060 0x4010e92e:0x3ffb0370 0x4010e96a:0x3ffb0400 0x400dba69:0x3ffb0440 0x400dba9d:0x3ffb0470 0x400dbad1:0x3ffb04c0 0x400dbc91:0x3ffb04e0 0x400e33d1:0x3ffb0500 0x400e33fe:0x3ffb0530 0x400df8b1:0x3ffb0590 0x401635d5:0x3ffb06a0 0x401636a1:0x3ffb06c0 0x400e266d:0x3ffb06e0 0x400e8efa:0x3ffb0730 0x400ed282:0x3ffb1fb0 0x40089b92:0x3ffb1fd0
[06:00:15]
[06:00:15]Rebooting...

syssi, 2022-05-22:
Do you use an ESP32 or ESP8266? Could you remove most of the sensors for testing? Just a few should remain.

syssi, 2022-05-22:
I've pushed a fix. Please try again:

https://github.com/syssi/esphome-jbd-bms/commit/130ce72cd313324fe87b1fa249114adc4dc1373e

I've changed the framework to `esp-idf` because there is something odd with the `arduino` framework of the most recent ESPHome version (I assume).

syssi, 2022-05-22:
If the component does a good job for you please report your BMS model name. I would like to extend the list of supported models. One important info: While developing the component using an ESP8266 I noticed the BMS doesn't respond if the request was send via hardware serial. I've changed the GPIOs to use software serial. In this case the BMS did understand/received my requests and did provide a response. Using `GPIO1` and `GPIO3` the BMS didn't respond (verified via logic analyzer). It's possible the BMS also doesn't like the hardware UART implementation of the ESP32. Please ping me if you see outgoind traffic but no responses.

Bretthansenpc, 2022-05-22:
I'll post as soon as I test it Thank You for the fast reply!

Bretthansenpc, 2022-05-22:
Model: JBD-SP04S020-L4S-120A-B-U-K

It works as far as showing all the voltages and such only thing is the current and power jump around oddly usually it shows everything correctly but jumps to 600+ on charge current and then on power, it jumps to 9000 watts.
[image]

syssi, 2022-05-22:
Could you provide some raw values (can be extracted from the logs if you increase the log level). I assume the (signed?) values aren't interpreted/castet properly. My JBD haven't seen any incoming/outgoing current yet. So it's very likely something is wrong here.

syssi, 2022-05-22:
Could you test this feature branch (fix-current-sensor) by updating your config yaml slightly:

```
substitutions:
  name: jbd-bms-uart
  external_components_source: github://syssi/esphome-jbd-bms@fix-current-sensor
```

The solution (https://github.com/syssi/esphome-jbd-bms/pull/2/files) is just a guess. If you provide some logs (including raw frames) it's easier for me to verify the solution locally.

Bretthansenpc, 2022-05-22:
Awesome I'll update and go from there when I get home thank you!

Bretthansenpc, 2022-05-23:
its still doing it but here's some of the logs I'll have to get more to actually show it I'm sure.

[https://pastebin.com/pg6QBp4p](url)

syssi, 2022-05-23:
Your log didn't help because it contains `0 A` current readings only. I need a log containg one of the `650.9 A` readings above.

syssi, 2022-05-23:
A log containing some load (discharging vs. charging) would be nice too. I'm unsure the JBD BMS is able to distinguish between "incoming" and "outgoing" current.

syssi, 2022-05-23:
I've attached some load to my JBD BMS and was able to fix the current & power sensor.

closed 2022-05-23 by syssi