syssi/esphome-jbd-bms #7: BLE Support

RoySalisbury, 2022-08-11:
Please add BLE support for this device...  

Thanks

syssi, 2022-08-12:
Implementation details: https://github.com/tgalarneau/bms/blob/main/jbdbms-4-socket-2temps.py#L180-L188

syssi, 2022-09-26:
- [ ] Record a btsnoop capture to get a better understanding of the BLE traffic
- [ ] Identify available and used services, chars and handles
- [ ] Subscribe to notifications
- [ ] Request data periodically (0x03: cell info, 0x04: cell voltages)
- [ ] Receive notifications and pass the data to the decoder

syssi, 2022-09-26:
I assume the BLE support is ready at the end of the week. I've made good progress today:

```
[19:30:56][I][jbd_bms_ble:135]: Request status notification
[19:30:56][D][jbd_bms_ble:385]: Write register: DD.A5.03.00.FF.FD.77 (7)
[19:30:56][D][jbd_bms_ble:100]: Notification received: DD.03.00.1D.05.59.00.00.6D.5E.6D.60.00.00.2C.7C.00.00.00.00 (20)
[19:30:56][D][jbd_bms_ble:100]: Notification received: 00.00.80.64.03.04.03.0B.6E.0B.6E.0B.6A.FA.F0.77 (16)
[19:30:58][I][jbd_bms_ble:135]: Request status notification
[19:30:58][D][jbd_bms_ble:385]: Write register: DD.A5.03.00.FF.FD.77 (7)
[19:30:58][D][jbd_bms_ble:100]: Notification received: DD.03.00.1D.05.59.00.00.6D.5E.6D.60.00.00.2C.7C.00.00.00.00 (20)
[19:30:58][D][jbd_bms_ble:100]: Notification received: 00.00.80.64.03.04.03.0B.6E.0B.6E.0B.6A.FA.F0.77 (16)
```

syssi, 2022-09-26:
A first draft is ready. Please give it a try:

https://github.com/syssi/esphome-jbd-bms/blob/add-ble-support/esp32-ble-example.yaml

Just adjust the `mac_address` and flash the BLE example configuration to an ESP32 near your JBD BMS.

RoySalisbury, 2022-09-26:
I have installed it to one of my ESP32's.  Seems to be working fine.  Great work.  I will let it continue to operate and let you know how it works out.

JeffyBeepBoop, 2022-09-27:
Tried it on a new d1 mini esp32 and I'm getting a compile error. This is the first time using the esp-idf framework. There were a couple download freezes along the way, retrying the install would seemingly allow the toolchain downloads to continue. 

`HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40302.0 (4.3.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch2
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/jbd-bms-ble/.pioenvs/jbd-bms-ble/CMakeFiles/CMakeOutput.log".
See also "/data/jbd-bms-ble/.pioenvs/jbd-bms-ble/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/jbd-bms-ble/.pioenvs/jbd-bms-ble/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_f5826 && [1/2] Building C object CMakeFiles/cmTC_f5826.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_f5826.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_f5826.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc: not found
    ninja: build stopped: subcommand failed.`

syssi, 2022-09-27:
The most critical issue is this one:

```
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```

It looks like the build folder and your home directory aren't on the same filesystem. Could you try to build the YAML at a working directory in your home directory? If your home directory is a symlink please make sure `echo $HOME` doesn't use the symlink.

```
[sebastian@leonard ~] echo $HOME
/home/sebastian
[sebastian@leonard ~] ls -l /home 
lrwxrwxrwx 1 root root 15 13. Dez 2018  /home -> /srv/raid5/home
[sebastian@leonard ~] export HOME=/srv/raid5/home/sebastian
[sebastian@leonard /home/sebastian]
```

PR #9 (2022-09-27): Add BLE support
Fixes #7

- [x] Record a btsnoop capture to get a better understanding of the BLE traffic
- [x] Identify available and used services, chars and handles
- [x] Subscribe to notifications
- [x] Request data periodically (0x03: cell info, 0x04: cell voltages)
- [x] Receive notifications and pass the data to the decoder

# Available BLE services, chars and handles

```
Found device at MAC address [70:3E:97:07:C0:3E]
 Attempting BLE connection to 70:3e:97:07:c0:3e
 Service UUID: 0x1800
   start_handle: 0x1  end_handle: 0x7
  characteristic 0x2A00, handle 0x3, properties 0x12
  charac [... cut]
closed 2022-09-27 by syssi