syssi/esphome-jbd-bms #14: JBD-SP04S020A-L4S-80A-B-U Invalid Header

kevin-jake, 2022-10-26:
Hi I have successfully integrated this on my JBD 4s 80A BMS however I am getting invalid header errors. Any idea how to fix this? Also it seems that the binary sensors (Balancing, Discharging and Charging) is not working?? Please see the sample logs I am having:

`[18:07:52][W][jbd_bms:124]: Invalid header
[18:07:53][D][jbd_bms:248]: Hardware info:
[18:07:53][D][jbd_bms:249]:   Device model: JBD-SP04S020A-L4S-80A-B-U
[18:07:53][D][sensor:126]: 'jbd-bms-uart total voltage': Sending state 12.86000 V with 2 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart current': Sending state -4.91000 A with 1 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart power': Sending state -63.14260 W with 1 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart charging power': Sending state 0.00000 W with 2 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart discharging power': Sending state 63.14260 W with 2 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart capacity remaining': Sending state 0.00000 Ah with 2 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart nominal capacity': Sending state 5.40000 Ah with 2 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart charging cycles': Sending state 4.00000  with 0 decimals of accuracy
[18:07:53][D][jbd_bms:265]:   Date of manufacture: 2021.12.18
[18:07:53][D][sensor:126]: 'jbd-bms-uart balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[18:07:53][D][text_sensor:067]: 'jbd-bms-uart errors': Sending state ''
[18:07:53][D][sensor:126]: 'jbd-bms-uart software version': Sending state 2.00000  with 1 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart battery strings': Sending state 4.00000  with 0 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart temperature 1': Sending state 28.60000 °C with 1 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart temperature 2': Sending state 27.60000 °C with 1 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart temperature 3': Sending state 27.30000 °C with 1 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart cell voltage 1': Sending state 3.21900 V with 3 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart cell voltage 2': Sending state 3.22100 V with 3 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart cell voltage 3': Sending state 3.20900 V with 3 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart cell voltage 4': Sending state 3.21900 V with 3 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart min cell voltage': Sending state 3.20900 V with 3 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart max cell voltage': Sending state 3.22100 V with 3 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart max voltage cell': Sending state 2.00000  with 0 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart min voltage cell': Sending state 3.00000  with 0 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart delta cell voltage': Sending state 0.01200 V with 4 decimals of accuracy
[18:07:53][D][sensor:126]: 'jbd-bms-uart average cell voltage': Sending state 3.21700 V with 4 decimals of accuracy
[18:07:55][D][jbd_bms:248]: Hardware info:
[18:07:55][D][jbd_bms:249]:   Device model: JBD-SP04S020A-L4S-80A-B-U
[18:07:55][D][sensor:126]: 'jbd-bms-uart total voltage': Sending state 12.86000 V with 2 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart current': Sending state -4.76000 A with 1 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart power': Sending state -61.21360 W with 1 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart charging power': Sending state 0.00000 W with 2 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart discharging power': Sending state 61.21360 W with 2 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart capacity remaining': Sending state 0.00000 Ah with 2 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart nominal capacity': Sending state 5.40000 Ah with 2 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart charging cycles': Sending state 4.00000  with 0 decimals of accuracy
[18:07:55][D][jbd_bms:265]:   Date of manufacture: 2021.12.18
[18:07:55][D][sensor:126]: 'jbd-bms-uart balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[18:07:55][D][text_sensor:067]: 'jbd-bms-uart errors': Sending state ''
[18:07:55][D][sensor:126]: 'jbd-bms-uart software version': Sending state 2.00000  with 1 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart battery strings': Sending state 4.00000  with 0 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart temperature 1': Sending state 28.60000 °C with 1 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart temperature 2': Sending state 27.60000 °C with 1 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart temperature 3': Sending state 27.30000 °C with 1 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart cell voltage 1': Sending state 3.21700 V with 3 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart cell voltage 2': Sending state 3.22100 V with 3 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart cell voltage 3': Sending state 3.20900 V with 3 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart cell voltage 4': Sending state 3.22100 V with 3 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart min cell voltage': Sending state 3.20900 V with 3 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart max cell voltage': Sending state 3.22100 V with 3 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart max voltage cell': Sending state 2.00000  with 0 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart min voltage cell': Sending state 3.00000  with 0 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart delta cell voltage': Sending state 0.01200 V with 4 decimals of accuracy
[18:07:55][D][sensor:126]: 'jbd-bms-uart average cell voltage': Sending state 3.21700 V with 4 decimals of accuracy
[18:07:57][D][jbd_bms:248]: Hardware info:
[18:07:57][D][jbd_bms:249]:   Device model: JBD-SP04S020A-L4S-80A-B-U
[18:07:57][D][sensor:126]: 'jbd-bms-uart total voltage': Sending state 12.86000 V with 2 decimals of accuracy
[18:07:57][D][sensor:126]: 'jbd-bms-uart current': Sending state -4.43000 A with 1 decimals of accuracy
[18:07:57][D][sensor:126]: 'jbd-bms-uart power': Sending state -56.96980 W with 1 decimals of accuracy
[... 23 lines cut ...]
[18:07:57][D][sensor:126]: 'jbd-bms-uart min voltage cell': Sending state 3.00000  with 0 decimals of accuracy
[18:07:57][D][sensor:126]: 'jbd-bms-uart delta cell voltage': Sending state 0.01000 V with 4 decimals of accuracy
[18:07:57][D][sensor:126]: 'jbd-bms-uart average cell voltage': Sending state 3.21550 V with 4 decimals of accuracy
[18:07:59][D][jbd_bms:248]: Hardware info:
[18:07:59][D][jbd_bms:249]:   Device model: JBD-SP04S020A-L4S-80A-B-U
[18:07:59][D][sensor:126]: 'jbd-bms-uart total voltage': Sending state 12.87000 V with 2 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart current': Sending state -4.43000 A with 1 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart power': Sending state -57.01410 W with 1 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart charging power': Sending state 0.00000 W with 2 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart discharging power': Sending state 57.01410 W with 2 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart capacity remaining': Sending state 0.00000 Ah with 2 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart nominal capacity': Sending state 5.40000 Ah with 2 decimals of accuracy
[... cut ...]
[18:07:59][D][sensor:126]: 'jbd-bms-uart operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart battery strings': Sending state 4.00000  with 0 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart temperature 1': Sending state 28.60000 °C with 1 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart temperature 2': Sending state 27.60000 °C with 1 decimals of accuracy
[18:07:59][D][sensor:126]: 'jbd-bms-uart temperature 3': Sending state 27.30000 °C with 1 decimals of accuracy
[18:07:59][W][jbd_bms:124]: Invalid header`

kevin-jake, 2022-10-26:
> Could you enable the debug output of the `uart` component:
> 
> ```
> logger:
>   level: DEBUG
> 
> uart:
>   id: uart0
>   baud_rate: 9600
>   tx_pin: GPIO4
>   rx_pin: GPIO5
>   debug:
>     direction: BOTH
> ```
> 
> and provide a log including multiple "Invalid header" warnings?

Got this on the logs:

```[21:03:16][D][uart_debug:114]: <<< FF
[21:03:16][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][W][jbd_bms:124]: Invalid header
[21:03:16][D][uart_debug:114]: <<< DD:03:00:1D:04:FC:FF:34:00:00:02:1C:00:05:2B:92:00:00:00:00:00:00:20:00:03:04:03:0B:CA:0B:C1:0B:BF:FA:3B:77
[21:03:16][D][uart_debug:114]: <<< FF
[21:03:18][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[21:03:18][D][jbd_bms:248]: Hardware info:
[21:03:18][D][jbd_bms:249]:   Device model: JBD-SP04S020A-L4S-80A-B-U
[21:03:18][D][sensor:126]: 'jbd-bms-uart total voltage': Sending state 12.75000 V with 2 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart current': Sending state -3.76000 A with 1 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart power': Sending state -47.94000 W with 1 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart charging power': Sending state 0.00000 W with 2 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart discharging power': Sending state 47.94000 W with 2 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart capacity remaining': Sending state 0.00000 Ah with 2 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart nominal capacity': Sending state 5.40000 Ah with 2 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart charging cycles': Sending state 5.00000  with 0 decimals of accuracy
[21:03:18][D][jbd_bms:265]:   Date of manufacture: 2021.12.18
[21:03:18][D][sensor:126]: 'jbd-bms-uart balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[21:03:18][D][text_sensor:067]: 'jbd-bms-uart errors': Sending state ''
[21:03:18][D][sensor:126]: 'jbd-bms-uart software version': Sending state 2.00000  with 1 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart battery strings': Sending state 4.00000  with 0 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart temperature 1': Sending state 28.70000 °C with 1 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart temperature 2': Sending state 27.80000 °C with 1 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart temperature 3': Sending state 27.60000 °C with 1 decimals of accuracy
[21:03:18][D][uart_debug:114]: <<< DD:03:00:1D:04:FB:FE:88:00:00:02:1C:00:05:2B:92:00:00:00:00:00:00:20:00:03:04:03:0B:CA:0B:C1:0B:BF:F9:E9:77
[21:03:18][D][uart_debug:114]: >>> DD:A5:04:00:FF:FC:77
[21:03:18][D][sensor:126]: 'jbd-bms-uart cell voltage 1': Sending state 3.19100 V with 3 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart cell voltage 2': Sending state 3.19300 V with 3 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart cell voltage 3': Sending state 3.18600 V with 3 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart cell voltage 4': Sending state 3.19300 V with 3 decimals of accuracy
[21:03:18][D][sensor:126]: 'jbd-bms-uart min cell voltage': Sending state 3.18600 V with 3 decimals of accuracy
[... 219 lines cut ...]
[21:03:34][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[21:03:34][D][jbd_bms:248]: Hardware info:
[21:03:34][D][jbd_bms:249]:   Device model: JBD-SP04S020A-L4S-80A-B-U
[21:03:34][D][sensor:126]: 'jbd-bms-uart total voltage': Sending state 12.77000 V with 2 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart current': Sending state -2.04000 A with 1 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart power': Sending state -26.05080 W with 1 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart charging power': Sending state 0.00000 W with 2 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart discharging power': Sending state 26.05080 W with 2 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart capacity remaining': Sending state 0.00000 Ah with 2 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart nominal capacity': Sending state 5.40000 Ah with 2 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart charging cycles': Sending state 5.00000  with 0 decimals of accuracy
[21:03:34][D][jbd_bms:265]:   Date of manufacture: 2021.12.18
[... cut ...]
[21:03:34][D][sensor:126]: 'jbd-bms-uart temperature 1': Sending state 28.70000 °C with 1 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart temperature 2': Sending state 27.80000 °C with 1 decimals of accuracy
[21:03:34][D][sensor:126]: 'jbd-bms-uart temperature 3': Sending state 27.60000 °C with 1 decimals of accuracy
[21:03:34][D][uart_debug:114]: <<< DD:03:00:1D:04:FD:FF:34:00:00:02:1C:00:05:2B:92:00:00:00:00:00:00:20:00:03:04:03:0B:CA:0B:C1:0B:BF:FA:3A:77
[21:03:34][D][uart_debug:114]: >>> DD:A5:04:00:FF:FC:77
[21:03:34][W][jbd_bms:147]: JbdBms CRC Check failed! FDF0 != F5F0

syssi, 2022-10-26:
Could you try to increase the `rx_timeout` from 150ms to 200ms and try again?

kevin-jake, 2022-10-26:
Invalid Headers occur less but still the Balancing, Charging and Discharging switches are still not working.

syssi, 2022-10-26:
Are we talking about the binary sensors or switches?

kevin-jake, 2022-10-26:
> switches
I increased timeout to 500ms. Looks fine now for the invalid header issue. Thanks for the tip man!

its the switches. I can see the commands being sent on the logs but its not taking effect. Also is it possible to change other settings for the BMS in home assistant?

syssi, 2022-10-26:
Switches aren't supported yet / they are untested: https://github.com/syssi/esphome-jbd-bms/blob/main/components/jbd_bms/switch/jbd_switch.cpp#L12

syssi, 2022-10-26:
No other controls are implemented yet.

kevin-jake, 2022-10-26:
Ok thanks for this if I got a chance to explore this maybe I'll just raise a PR but still need to learn the registers and how it communicates to the BMS.

Thank you for the great work!

syssi, 2022-10-26:
You could use this implementation / skeleton for some guidance to implement additional entities (number, button, etc.): https://github.com/syssi/esphome-jk-bms/tree/main/components/jk_bms_ble

I recommend to extract the commands/registers from a btsnoop capture to be sure to not pass garbage to the BMS.

kevin-jake, 2022-10-27:
Hi Invalid Headers occured again today and became more frequent, is there something i should do like resetting the bms or something?

```[19:28:34][W][jbd_bms:124]: Invalid header
[19:28:34][W][jbd_bms:124]: Invalid header
[19:28:34][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:DD:03:00:1D:05:15:FC:92:00:00:02:1C:00:0B:2B:92:00:00:00:00:00:00:20:00:03:04:03:0B:DB:0B:CD:0B:CB:FA:97:77:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:FF
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][W][jbd_bms:124]: Invalid header
[19:28:35][D][uart_debug:114]: <<< FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][D][uart_debug:114]: <<< FF
[19:28:36][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][D][uart_debug:114]: <<< FF:DD:03:00:1D:05:16:FD:15:00:00:02:1C:00:0B:2B:92:00:00:00:00:00:00:20:00:03:04:03:0B:DB:0B:CD:0B:CB:FB:12:77:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][D][uart_debug:114]: <<< FF:FF
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header
[19:28:36][W][jbd_bms:124]: Invalid header

syssi, 2022-10-27:
Another user did report in the past the serial interface of the BMS is unstable in case of charging the battery.

kevin-jake, 2022-10-27:
oh ok so this is a known issue. Perhaps in the BMS itself?

syssi, 2022-10-27:
Yes. If you have a logic analyzer you could try to record the traffic while charging & discharging. I assume you will see gaps/some silence between the bytes of a frame depending on the "load".

syssi, 2022-12-18:
Let's close this issue because its a hardware issue probably and cannot be solved here.

closed 2022-12-18 by syssi