syssi/esphome-jbd-bms #20: DP04S007 100A bluetooth

reflexsa, 2022-12-16:
First of all, thank you very much for making this integration.
I'm attempting to connect to my BMS (model from xiaoxiang app: DP04S007L4S100A, so probably translates to DP04S007 100A) via bluetooth using an ESP32WROOM.
I'm using the "esp32-ble-example.yaml" example, only changing the board to esp32dev and the MAC to that of my BMS. Once I have 'installed' the code onto the ESP, the log shows "Not connected" with the MAC address I provided. I assume this MAC address is that of the BMS?
I'm not sure if I am doing something wrong or if this BMS may not be supported.
Any advice is appreciated.

syssi, 2022-12-16:
You are on the right track. You have to make sure the MAC address matches with the one of your BMS. That's all.

Please increase the log level to `VERY_VERBOSE`:

```
logger:
  level: VERY_VERBOSE
```

Please reflash the ESPHome node and provida a log showing the "No connected" issue. I would like to make sure the implementation is able to see your BMS in the air and tries to connect to the BMS (without success?).

reflexsa, 2022-12-17:
I swapped out my ESP as I started to suspect that one has some signal/antenna issues.
Results look more promising, Note that I just changed the MAC address as I'm not sure what is safe to share.
Normal logger
`[08:08:40][C][jbd_bms_ble:384]: Cell Voltage 32 'revov-bms cell voltage 32'
[08:08:40][C][jbd_bms_ble:384]:   Device Class: 'voltage'
[08:08:40][C][jbd_bms_ble:384]:   State Class: 'measurement'
[08:08:40][C][jbd_bms_ble:384]:   Unit of Measurement: 'V'
[08:08:40][C][jbd_bms_ble:384]:   Accuracy Decimals: 3
[08:08:40][W][jbd_bms_ble:172]: [AA:AA:AA:AA:AA:AA] Not connected
[08:08:40][C][ble_switch:076]: BLE Client Switch 'revov-bms enable bluetooth connection'
[08:08:40][C][ble_switch:078]:   Icon: 'mdi:bluetooth'
[08:08:40][C][ble_switch:099]:   Restore Mode: restore defaults to OFF
[08:08:40][C][jbd_bms_ble.switch:076]: JbdBmsBle Switch 'revov-bms discharging'
[08:08:40][C][jbd_bms_ble.switch:078]:   Icon: 'mdi:battery-charging-50'
[08:08:40][C][jbd_bms_ble.switch:099]:   Restore Mode: restore defaults to OFF
[08:08:40][C][jbd_bms_ble.switch:076]: JbdBmsBle Switch 'revov-bms charging'
[08:08:40][C][jbd_bms_ble.switch:078]:   Icon: 'mdi:battery-charging-50'
[08:08:40][C][jbd_bms_ble.switch:099]:   Restore Mode: restore defaults to OFF
[08:08:40][C][jbd_bms_ble.switch:076]: JbdBmsBle Switch 'revov-bms balancer'
[08:08:40][C][jbd_bms_ble.switch:078]:   Icon: 'mdi:seesaw'
[08:08:40][C][jbd_bms_ble.switch:099]:   Restore Mode: restore defaults to OFF
[08:08:40][C][esp32_ble_tracker:874]: BLE Tracker:
[08:08:40][C][esp32_ble_tracker:875]:   Scan Duration: 300 s
[08:08:40][C][esp32_ble_tracker:876]:   Scan Interval: 320.0 ms
[08:08:40][C][esp32_ble_tracker:877]:   Scan Window: 30.0 ms
[08:08:40][C][esp32_ble_tracker:878]:   Scan Type: ACTIVE
[08:08:40][C][esp32_ble_tracker:879]:   Continuous Scanning: True
[08:08:40][C][ble_client:027]: BLE Client:
[08:08:40][C][ble_client:028]:   Address: AA:AA:AA:AA:AA:AA
[08:08:40][C][mdns:103]: mDNS:
[08:08:40][C][mdns:104]:   Hostname: revov-bms
[08:08:40][C][ota:093]: Over-The-Air Updates:
[08:08:40][C][ota:094]:   Address: 192.1.1.1:3232
[08:08:40][C][ota:097]:   Using Password.
[08:08:40][C][api:138]: API Server:
[08:08:40][C][api:139]:   Address: 192.1.1.1:6053
[08:08:40][C][api:143]:   Using noise encryption: NO
[08:08:42][W][jbd_bms_ble:172]: [AA:AA:AA:AA:AA:AA] Not connected
[08:08:44][W][jbd_bms_ble:172]: [AA:AA:AA:AA:AA:AA] Not connected
[08:08:45][D][esp32_ble_client:048]: [0] [AA:AA:AA:AA:AA:AA] Found device
[08:08:45][D][esp32_ble_tracker:216]: Pausing scan to make connection...
[08:08:45][I][esp32_ble_client:064]: [0] [AA:AA:AA:AA:AA:AA] 0x00 Attempting BLE connection
[08:08:46][W][jbd_bms_ble:172]: [AA:AA:AA:AA:AA:AA] Not connected
[08:08:46][D][api:102]: Accepted ::FFFF:192.1.1.2
[08:08:46][W][api.connection:083]: Home Assistant 2022.12.6 (::FFFF:192.1.1.2): Connection closed
[08:08:47][D][esp32_ble_tracker:327]: Starting scan...
[08:08:47][I][jbd_bms_ble:091]: Request device info
[08:08:48][W][jbd_bms_ble:114]: Maximum response size exceeded
[08:08:48][W][jbd_bms_ble:132]: Invalid frame length
[08:08:50][W][jbd_bms_ble:114]: Maximum response size exceeded
[08:08:50][W][jbd_bms_ble:132]: Invalid frame length`

Very verbose:
`[07:53:31][C][jbd_bms_ble:384]: Cell Voltage 32 'revov-bms cell voltage 32'
[07:53:31][C][jbd_bms_ble:384]:   Device Class: 'voltage'
[07:53:31][C][jbd_bms_ble:384]:   State Class: 'measurement'
[07:53:31][C][jbd_bms_ble:384]:   Unit of Measurement: 'V'
[07:53:31][C][jbd_bms_ble:384]:   Accuracy Decimals: 3
[07:53:31][VV][scheduler:196]: Running interval 'update' with interval=2000 last_execution=7232 (now=9659)
[07:53:31][VV][jbd_bms_ble:425]: Send command: DD.A5.03.00.FF.FD.77 (7)
[07:53:31][C][ble_switch:076]: BLE Client Switch 'revov-bms enable bluetooth connection'
[07:53:31][C][ble_switch:078]:   Icon: 'mdi:bluetooth'
[07:53:31][C][ble_switch:099]:   Restore Mode: restore defaults to OFF
[07:53:31][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=4 gattc_if=3
[07:53:31][C][jbd_bms_ble.switch:076]: JbdBmsBle Switch 'revov-bms discharging'
[07:53:31][C][jbd_bms_ble.switch:078]:   Icon: 'mdi:battery-charging-50'
[07:53:31][C][jbd_bms_ble.switch:099]:   Restore Mode: restore defaults to OFF
[07:53:31][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=10 gattc_if=3
[07:53:31][VV][jbd_bms_ble:101]: Notification received: DD.03.00.22.05.5F.00.00.4A.DF.4E.20.00.02.2D.14.00.00.00.00 (20)
[07:53:31][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=10 gattc_if=3
[... 140 lines cut ...]
[07:53:41][W][jbd_bms_ble:132]: Invalid frame length
[07:53:43][VV][scheduler:196]: Running interval 'update' with interval=2000 last_execution=19232 (now=21232)
[07:53:43][VV][jbd_bms_ble:425]: Send command: DD.A5.03.00.FF.FD.77 (7)
[07:53:43][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=4 gattc_if=3
[07:53:43][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=10 gattc_if=3
[07:53:43][VV][jbd_bms_ble:101]: Notification received: DD.03.00.22.05.5F.00.00.4A.DF.4E.20.00.02.2D.14.00.00.00.00 (20)
[07:53:43][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=10 gattc_if=3
[07:53:43][VV][jbd_bms_ble:101]: Notification received: 00.00.23.60.03.04.01.0B.B1.00.00.00.4E.20.4A.DF.00.00.FA.C2 (20)
[... cut ...]
[07:53:45][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=10 gattc_if=3
[07:53:45][VV][jbd_bms_ble:101]: Notification received: 00.00.23.60.03.04.01.0B.B1.00.00.00.4E.20.4A.DF.00.00.FA.C2 (20)
[07:53:45][V][esp32_ble_client:111]: [0] [AA:AA:AA:AA:AA:AA] gattc_event_handler: event=10 gattc_if=3
[07:53:45][VV][jbd_bms_ble:101]: Notification received: 77
[07:53:45][W][jbd_bms_ble:114]: Maximum response size exceeded
[07:53:45][W][jbd_bms_ble:132]: Invalid frame length`

syssi, 2022-12-17:
Good job! I will extract the responses from the log now and try to get an idea what needs to be improved to support your model. In best case relaxing some upper limits is enough.

syssi, 2022-12-17:
Please change your configuration yaml to this feature branch and compile + flash again:

``
substitutions:
  external_components_source: github://syssi/esphome-jbd-bms@add-dp04s007-support
```

I've increased the `MAX_RESPONSE_SIZE` to not reject the 41 byte frames of your BMS anymore:

https://github.com/syssi/esphome-jbd-bms/pull/21/files

reflexsa, 2022-12-17:
Looks like that did the trick! Thank you so much for such speedy assistance. Not seeing my device/entities yet under integrations but that is something else.

10:49:48][D][esp32_ble_tracker:327]: Starting scan...
[10:49:48][I][jbd_bms_ble:091]: Request device info
[10:49:48][D][jbd_bms_ble:266]: Hardware info:
[10:49:48][D][jbd_bms_ble:267]:   Device model: 
[10:49:48][D][sensor:127]: 'revov-bms total voltage': Sending state 13.28000 V with 2 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms current': Sending state -4.88000 A with 1 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms power': Sending state -64.80640 W with 1 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms charging power': Sending state 0.00000 W with 2 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms discharging power': Sending state 64.80640 W with 2 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms capacity remaining': Sending state 190.14000 Ah with 2 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms nominal capacity': Sending state 200.00000 Ah with 2 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms charging cycles': Sending state 2.00000  with 0 decimals of accuracy
[10:49:48][D][jbd_bms_ble:284]:   Date of manufacture: 2022.8.20
[10:49:48][D][sensor:127]: 'revov-bms balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[10:49:48][D][binary_sensor:036]: 'revov-bms balancing': Sending state OFF
[10:49:48][D][sensor:127]: 'revov-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[10:49:48][D][text_sensor:067]: 'revov-bms errors': Sending state ''
[10:49:48][D][sensor:127]: 'revov-bms software version': Sending state 2.30000  with 1 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms state of charge': Sending state 95.00000 % with 0 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[10:49:48][D][binary_sensor:036]: 'revov-bms charging': Sending state ON
[10:49:48][D][binary_sensor:036]: 'revov-bms discharging': Sending state ON
[10:49:48][D][sensor:127]: 'revov-bms battery strings': Sending state 4.00000  with 0 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms temperature 1': Sending state 25.80000 °C with 1 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms cell voltage 1': Sending state 3.32000 V with 3 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms cell voltage 2': Sending state 3.31900 V with 3 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms cell voltage 3': Sending state 3.31900 V with 3 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms cell voltage 4': Sending state 3.32200 V with 3 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms min cell voltage': Sending state 3.31900 V with 3 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms max cell voltage': Sending state 3.32200 V with 3 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms max voltage cell': Sending state 4.00000  with 0 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms min voltage cell': Sending state 2.00000  with 0 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms delta cell voltage': Sending state 0.00300 V with 4 decimals of accuracy
[10:49:48][D][sensor:127]: 'revov-bms average cell voltage': Sending state 3.32000 V with 4 decimals of accuracy
[10:49:50][D][jbd_bms_ble:266]: Hardware info:
[10:49:50][D][jbd_bms_ble:267]:   Device model: 
[10:49:50][D][sensor:127]: 'revov-bms total voltage': Sending state 13.28000 V with 2 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms current': Sending state -4.88000 A with 1 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms power': Sending state -64.80640 W with 1 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms charging power': Sending state 0.00000 W with 2 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms discharging power': Sending state 64.80640 W with 2 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms capacity remaining': Sending state 190.12999 Ah with 2 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms nominal capacity': Sending state 200.00000 Ah with 2 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms charging cycles': Sending state 2.00000  with 0 decimals of accuracy
[10:49:50][D][jbd_bms_ble:284]:   Date of manufacture: 2022.8.20
[10:49:50][D][sensor:127]: 'revov-bms balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[10:49:50][D][text_sensor:067]: 'revov-bms errors': Sending state ''
[10:49:50][D][sensor:127]: 'revov-bms software version': Sending state 2.30000  with 1 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms state of charge': Sending state 95.00000 % with 0 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms battery strings': Sending state 4.00000  with 0 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms temperature 1': Sending state 25.80000 °C with 1 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms cell voltage 1': Sending state 3.32000 V with 3 decimals of accuracy
[... cut ...]
[10:49:50][D][sensor:127]: 'revov-bms max cell voltage': Sending state 3.32200 V with 3 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms max voltage cell': Sending state 4.00000  with 0 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms min voltage cell': Sending state 2.00000  with 0 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms delta cell voltage': Sending state 0.00300 V with 4 decimals of accuracy
[10:49:50][D][sensor:127]: 'revov-bms average cell voltage': Sending state 3.32000 V with 4 decimals of accuracy

syssi, 2022-12-17:
You are welcome! Could you do me a favor? This implementation sends three different requests to the BMS:

1. `>>> DD:A5:03:00:FF:FD:77`
2. `>>> DD:A5:04:00:FF:FC:77`
3. `>>> DD:A5:05:00:FF:FB:77`

You've provided an response example for request 1 already. I'm looking for example of type 2 and 3. I've pushed some more debug statements to the feature branch. Could you compile & flash the node again. This time you should see some raw traffic again:

```
Cell info frame (%d bytes): [...]
Hardware info frame (%d bytes): [...]
Hardware version frame (%d bytes): [...]
```
https://github.com/syssi/esphome-jbd-bms/pull/21/files

Could you provide these raw frames? Thanks!

syssi, 2022-12-17:
If you have used this ESP32 in the past as ESPHome node already with a different hostname you have to remove the old node (host)name from Home Assistant first. If you add the new ESPHome device afterwards the device + all entities should show up properly.

reflexsa, 2022-12-17:
Did not see the hardware version frame showing up

[15:16:55][D][jbd_bms_ble:251]: Hardware info frame (34 bytes):
[15:16:55][D][jbd_bms_ble:252]:   05.5E.00.00.4B.4D.4E.20.00.02.2D.14.00.00.00.00.00.00.23.60.03.04.01.0B.AF.00.00.00.4E.20.4B.4D.00.00 (34)
[15:16:55][D][jbd_bms_ble:272]: Hardware info:
[15:16:55][D][jbd_bms_ble:273]:   Device model: 
[15:16:55][D][sensor:127]: 'revov-bms total voltage': Sending state 13.74000 V with 2 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms current': Sending state 0.00000 A with 1 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms power': Sending state 0.00000 W with 1 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms charging power': Sending state 0.00000 W with 2 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms capacity remaining': Sending state 192.76999 Ah with 2 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms nominal capacity': Sending state 200.00000 Ah with 2 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms charging cycles': Sending state 2.00000  with 0 decimals of accuracy
[15:16:55][D][jbd_bms_ble:290]:   Date of manufacture: 2022.8.20
[15:16:55][D][sensor:127]: 'revov-bms balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[15:16:55][D][binary_sensor:036]: 'revov-bms balancing': Sending state OFF
[15:16:55][D][sensor:127]: 'revov-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[15:16:55][D][text_sensor:067]: 'revov-bms errors': Sending state ''
[15:16:55][D][sensor:127]: 'revov-bms software version': Sending state 2.30000  with 1 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms state of charge': Sending state 96.00000 % with 0 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms battery strings': Sending state 4.00000  with 0 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms temperature 1': Sending state 26.00000 °C with 1 decimals of accuracy
[15:16:55][D][jbd_bms_ble:200]: Cell info frame (8 bytes):
[15:16:55][D][jbd_bms_ble:201]:   0D.52.0D.56.0D.57.0D.B4 (8)
[15:16:55][D][sensor:127]: 'revov-bms cell voltage 1': Sending state 3.41000 V with 3 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms cell voltage 2': Sending state 3.41400 V with 3 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms cell voltage 3': Sending state 3.41500 V with 3 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms cell voltage 4': Sending state 3.50800 V with 3 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms min cell voltage': Sending state 3.41000 V with 3 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms max cell voltage': Sending state 3.50800 V with 3 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms max voltage cell': Sending state 4.00000  with 0 decimals of accuracy
[15:16:55][D][sensor:127]: 'revov-bms min voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[15:16:56][D][sensor:127]: 'revov-bms delta cell voltage': Sending state 0.09800 V with 4 decimals of accuracy
[15:16:56][D][sensor:127]: 'revov-bms average cell voltage': Sending state 3.43675 V with 4 decimals of accuracy
[15:16:57][D][jbd_bms_ble:251]: Hardware info frame (34 bytes):
[15:16:57][D][jbd_bms_ble:252]:   05.5E.00.00.4B.4D.4E.20.00.02.2D.14.00.00.00.00.00.00.23.60.03.04.01.0B.AF.00.00.00.4E.20.4B.4D.00.00 (34)
[15:16:57][D][jbd_bms_ble:272]: Hardware info:
[15:16:57][D][jbd_bms_ble:273]:   Device model: 
[15:16:57][D][sensor:127]: 'revov-bms total voltage': Sending state 13.74000 V with 2 decimals of accuracy

syssi, 2022-12-17:
The `hardware version frame` is requested once if the BLE connection is established. Please use the

```
switch:
  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"
```

switch to turn off the BLE connection + turn it on again. This time you should see the `hardware version frame`  at the log.

reflexsa, 2022-12-17:
I don't seem to see that frame when turning off and on the ble connection. 
When I turn off I get this:

[16:14:51][D][switch:017]: 'Revov enable bluetooth connection' Turning OFF.
[16:14:51][I][ble_client:041]: [AA:AA:AA:AA:AA:AA] Disabling BLE client.
[16:14:51][D][switch:056]: 'Revov enable bluetooth connection': Sending state OFF
[16:14:51][D][esp-idf:000]: W (1866637) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[16:14:51][D][esp-idf:000]: W (1866640) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[16:14:51][D][esp-idf:000]: W (1866683) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

When turning back on it seems to start with hardware info frame followed by cell info frame but no sign of the version frame.

syssi, 2022-12-17:
Alright. It looks like your BMS doesn't support this response. Thanks for your support! I will merge the feature branch into `main` now. Please don't forget to update your YAML.

syssi, 2022-12-17:
Feel free to close this issue if it's resolved.

reflexsa, 2022-12-17:
> 

Thanks so much for the help!

closed 2022-12-17 by reflexsa

droserHA, 2024-10-26:
thanks for your work. Everything worked for me. I added the status led and total energy charged and total energy discharged, daily energy charged and daily energy discharged