syssi/esphome-jbd-bms #23: how to get values (MQTT subscribe?) and store into DB

taHC81, 2023-01-24:
Hello, first, big thanks for this project, working great. I'd like to store data from a BMS to a DB (already used for inverter data), but struggling how to do that. I did a scan of MQTT topics with MQTT.fx tool, and if I subscribe to f.i. **homeassistant/sensor/jbd-bms-uart/jbd-bms-uart_discharging_power/config**, payload data are like below, but no value. How can I get them? Thanks!

`{"dev_cla":"power","unit_of_meas":"W","stat_cla":"measurement","name":"jbd-bms-uart discharging power","stat_t":"jbd-bms-uart/sensor/jbd-bms-uart_discharging_power/state","avty_t":"jbd-bms-uart/status","uniq_id":"ESPsensorjbd-bms-uart_discharging_power","dev":{"ids":"c8c9a30bf3fd","name":"jbd-bms-uart","sw":"esphome v2022.12.4 Jan 22 2023, 18:37:42","mdl":"nodemcu","mf":"espressif"}}`

syssi, 2023-01-24:
If your BMS is attached the component should request measurements periodically and publish these values as sensor `state`. You have to subscribe to all `state` topics you are interested in. You could use a tool `telegraf` to pass the data into a influxdb.

taHC81, 2023-01-24:
I had to enable mqtt in the yaml.

closed 2023-01-24 by taHC81