syssi/esphome-jbd-bms #25: Can't connect via ble version

yarosman, 2023-02-06:
Hello. I try to use default ble example with logging to connect JBD-AP21S001-L20S-150A-B, but nothing work

[02:03:56][VV][api.service:470]: on_ping_request: PingRequest {}
[02:03:56][VV][api.service:043]: send_ping_response: PingResponse {}
[02:03:56][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[02:03:58][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[02:04:00][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[02:04:02][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[02:04:04][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[02:04:06][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[02:04:08][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[02:04:10][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected

syssi, 2023-02-06:
Does your ESP32 report/see other BLE devices around you? Is your ESP able to see the BMS but unable to connect? How old/new is your BMS? Could you provide a photo of the BLE dongle? Do you own an Android device? I could provide some instructions how to do a traffic capture of the data between the Android App and the BMS.

yarosman, 2023-02-06:
- Does your ESP32 report/see other BLE devices around you?
Yes, I enable `esp32_ble_tracker`, `bluetooth_proxy` and saw other devices in the logs
- Is your ESP able to see the BMS but unable to connect? 
ESP was saw BMS because of 
```
[14:28:37][V][bluetooth_proxy:033]: Proxying packet from yarosman BMS - A4:C1:37:41:59:DA. RSSI: -55 dB
// after enabling jbd_bms_ble
[14:36:55][D][esp32_ble_tracker:216]: Pausing scan to make connection...
[14:36:55][I][esp32_ble_client:064]: [3] [A4:C1:37:41:59:DA] 0x00 Attempting BLE connection
[14:36:56][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
```
- How old/new is your BMS?
Manufact.data 10 Jan 2022
- Could you provide a photo of the BLE dongle? 
No, it's under the hood of my battery
- Do you own an Android device? I could provide some instructions how to do a traffic capture of the data between the Android App and the BMS.
No, I don't have, but I can try to get it

syssi, 2023-02-06:
Could you temporary remove the `bluetooth_proxy` component and tell me the ESPHome version you are using? A full log would be helpful too.

yarosman, 2023-02-06:
esphome version = 2022.12.8
<details><summary>Logs</summary>

[21:29:43][C][wifi:504]: WiFi:
[21:29:43][C][wifi:362]:   Local MAC: 78:21:84:BB:43:7C
[21:29:43][C][wifi:363]:   SSID: [redacted]
[21:29:43][C][wifi:364]:   IP Address: 192.168.1.208
[21:29:43][C][wifi:366]:   BSSID: [redacted]
[21:29:43][C][wifi:367]:   Hostname: 'jbd-bms'
[21:29:43][C][wifi:369]:   Signal strength: -32 dB ▂▄▆█
[21:29:43][V][wifi:371]:   Priority: 0.0
[21:29:43][C][wifi:373]:   Channel: 10
[21:29:43][C][wifi:374]:   Subnet: 255.255.255.0
[21:29:43][C][wifi:375]:   Gateway: 192.168.1.1
[21:29:43][C][wifi:376]:   DNS1: 192.168.1.1
[21:29:43][C][wifi:377]:   DNS2: 0.0.0.0
[21:29:43][C][logger:293]: Logger:
[21:29:43][C][logger:294]:   Level: VERY_VERBOSE
[21:29:43][C][logger:295]:   Log Baud Rate: 115200
[21:29:43][C][logger:296]:   Hardware UART: UART0
[21:29:43][C][logger:298]:   Level for 'scheduler': DEBUG
[21:29:43][C][logger:298]:   Level for 'component': DEBUG
[21:29:43][C][logger:298]:   Level for 'esp32_ble_tracker': DEBUG
[21:29:43][C][logger:298]:   Level for 'sensor': DEBUG
[21:29:43][C][jbd_bms_ble:337]: JbdBmsBle:
[21:29:43][C][jbd_bms_ble:338]:   Fake traffic enabled: NO
[21:29:43][C][jbd_bms_ble:340]: Total voltage 'jbd-bms total voltage'
[21:29:43][C][jbd_bms_ble:340]:   Device Class: 'voltage'
[21:29:43][C][jbd_bms_ble:340]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:340]:   Unit of Measurement: 'V'
[21:29:43][C][jbd_bms_ble:340]:   Accuracy Decimals: 2
[21:29:43][C][jbd_bms_ble:341]: Battery strings 'jbd-bms battery strings'
[21:29:43][C][jbd_bms_ble:341]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:341]:   Unit of Measurement: ''
[21:29:43][C][jbd_bms_ble:341]:   Accuracy Decimals: 0
[21:29:43][C][jbd_bms_ble:341]:   Icon: 'mdi:car-battery'
[21:29:43][C][jbd_bms_ble:342]: Software version 'jbd-bms software version'
[21:29:43][C][jbd_bms_ble:342]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:342]:   Unit of Measurement: ''
[21:29:43][C][jbd_bms_ble:342]:   Accuracy Decimals: 1
[21:29:43][C][jbd_bms_ble:343]: Current 'jbd-bms current'
[21:29:43][C][jbd_bms_ble:343]:   Device Class: 'current'
[21:29:43][C][jbd_bms_ble:343]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:343]:   Unit of Measurement: 'A'
[21:29:43][C][jbd_bms_ble:343]:   Accuracy Decimals: 1
[21:29:43][C][jbd_bms_ble:343]:   Icon: 'mdi:current-dc'
[21:29:43][C][jbd_bms_ble:344]: Power 'jbd-bms power'
[21:29:43][C][jbd_bms_ble:344]:   Device Class: 'power'
[21:29:43][C][jbd_bms_ble:344]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:344]:   Unit of Measurement: 'W'
[21:29:43][C][jbd_bms_ble:344]:   Accuracy Decimals: 1
[21:29:43][C][jbd_bms_ble:345]: Charging Power 'jbd-bms charging power'
[21:29:43][C][jbd_bms_ble:345]:   Device Class: 'power'
[21:29:43][C][jbd_bms_ble:345]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:345]:   Unit of Measurement: 'W'
[21:29:43][C][jbd_bms_ble:345]:   Accuracy Decimals: 2
[21:29:43][C][jbd_bms_ble:346]: Discharging Power 'jbd-bms discharging power'
[21:29:43][C][jbd_bms_ble:346]:   Device Class: 'power'
[21:29:43][C][jbd_bms_ble:346]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:346]:   Unit of Measurement: 'W'
[21:29:43][C][jbd_bms_ble:346]:   Accuracy Decimals: 2
[21:29:43][C][jbd_bms_ble:347]: State of charge 'jbd-bms state of charge'
[21:29:43][C][jbd_bms_ble:347]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:347]:   Unit of Measurement: '%'
[21:29:43][C][jbd_bms_ble:347]:   Accuracy Decimals: 0
[21:29:43][C][jbd_bms_ble:347]:   Icon: 'mdi:battery-50'
[21:29:43][C][jbd_bms_ble:348]: Nominal capacity 'jbd-bms nominal capacity'
[21:29:43][C][jbd_bms_ble:348]:   State Class: 'measurement'
[21:29:43][C][jbd_bms_ble:348]:   Unit of Measurement: 'Ah'
[21:29:43][C][jbd_bms_ble:348]:   Accuracy Decimals: 2
[... 262 lines cut ...]
[21:29:45][C][api:139]:   Address: jbd-bms.local:6053
[21:29:45][C][api:143]:   Using noise encryption: NO
[21:29:46][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:29:48][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:29:50][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:29:52][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:29:54][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:29:56][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:29:58][VV][api.service:470]: on_ping_request: PingRequest {}
[21:29:58][VV][api.service:043]: send_ping_response: PingResponse {}
[21:29:58][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:00][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:02][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:04][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:06][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:08][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:10][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:12][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:13][VV][api.service:470]: on_ping_request: PingRequest {}
[21:30:13][VV][api.service:043]: send_ping_response: PingResponse {}
[21:30:14][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:16][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected
[21:30:18][W][jbd_bms_ble:172]: [A4:C1:37:41:59:DA] Not connected

</details>

syssi, 2023-02-11:
Hmm..  we should capture some BLE traffic to get an idea what's different to my implementation. One last idea: May be the ESP is too close or to far away from your BMS. Play around with the distance.

yarosman, 2023-02-11:
@syssi Hello. I found the problem. There is some problem with jbd ble dongle, after some time it refuses to accept new connections, it was described in [batmon-ha](https://github.com/fl4p/batmon-ha#known-issues)

When I turned off Bluetooth on my phone and turned off/on jbd dongle, after that esp32 was connected to jbd

syssi, 2023-02-11:
Yes! The switches aren't implemented yet but they are coming soon!

closed 2023-02-11 by syssi