syssi/esphome-jbd-bms #28: bluetooth connection?

mager33, 2023-02-14:
Hi, does this work with a serial connection via bluetooth?

syssi, 2023-02-14:
This implementation supports both. A serial connection and communication via bluetooth.

syssi, 2023-02-15:
Does this answer your question?

mager33, 2023-02-15:
Yes, thank you! (unfortunately, I've discovered that my battery pack from BASEN does not contain a JDB BMS but a VIP BMS; and ther is not much available for that)

closed 2023-02-15 by mager33

syssi, 2023-02-15:
@mager33 I would like to give it a try to talk to your BMS. Do you have some spare time for doing some testing and experiments?

mager33, 2023-02-16:
Hi Sebastian, yes, of course. But I won't be available until 22.2.23. Best, Martin
 

Gesendet: Mittwoch, 15. Februar 2023 um 21:38 Uhr

syssi, 2023-02-16:
@mager33 To close the loop: We talked already on Reddit and you did provide the capture. :-P