syssi/esphome-jbd-bms #31: Overkill JBD-SP25S003-L16S-100A-100A-B support

Uksa007, 2023-03-09:
Hi,

I have your code working with the Device model: JBD-SP25S003-L16S-100A-100A-B
It works pretty well, apart from a few missing values, is there a possibly to resolve these issues?:

Operation status Unknown
Temperature 1 Unknown
Temperature 2 Unknown
Temperature 3 Unknown
Temperature 4 Unknown
Temperature 5 Unknown
Temperature 6 Unknown

With UART debug on I get the following:
```
[12:32:52][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[12:32:52][D][jbd_bms:248]: Hardware info:
[12:32:52][D][jbd_bms:249]:   Device model: JBD-SP25S003-L16S-100A-100A-B
[12:32:52][D][sensor:127]: 'jbd-bms-uart total voltage': Sending state 0.00000 V with 2 decimals of accuracy
[12:32:52][D][sensor:127]: 'jbd-bms-uart current': Sending state 0.00000 A with 1 decimals of accuracy
[12:32:52][D][sensor:127]: 'jbd-bms-uart power': Sending state 0.00000 W with 1 decimals of accuracy
[12:32:52][D][sensor:127]: 'jbd-bms-uart charging power': Sending state 0.00000 W with 2 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart capacity remaining': Sending state 0.00000 Ah with 2 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart nominal capacity': Sending state 100.00000 Ah with 2 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart charging cycles': Sending state 0.00000  with 0 decimals of accuracy
[12:32:53][D][jbd_bms:266]:   Date of manufacture: 2022.2.16
[12:32:53][D][sensor:127]: 'jbd-bms-uart balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[12:32:53][D][text_sensor:067]: 'jbd-bms-uart errors': Sending state ''
[12:32:53][D][sensor:127]: 'jbd-bms-uart software version': Sending state 2.00000  with 1 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart operation status bitmask': Sending state 1.00000  with 0 decimals of accuracy
[12:32:53][D][sensor:127]: 'jbd-bms-uart battery strings': Sending state 16.00000  with 0 decimals of accuracy
[12:32:53][D][uart_debug:114]: <<< DD:03:00:17:00:00:00:00:00:00:27:10:00:00:2C:50:00:00:00:00:00:00:20:00:01:10:00:FF:05:77
[12:32:53][D][uart_debug:114]: >>> DD:A5:04:00:FF:FC:77
[12:32:54][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[12:32:56][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[12:32:56][D][jbd_bms:248]: Hardware info:
[12:32:56][D][jbd_bms:249]:   Device model: JBD-SP25S003-L16S-100A-100A-B
[12:32:56][D][sensor:127]: 'jbd-bms-uart total voltage': Sending state 0.00000 V with 2 decimals of accuracy
[12:32:56][D][sensor:127]: 'jbd-bms-uart current': Sending state 0.00000 A with 1 decimals of accuracy
[12:32:56][D][sensor:127]: 'jbd-bms-uart power': Sending state 0.00000 W with 1 decimals of accuracy
[12:32:56][D][sensor:127]: 'jbd-bms-uart charging power': Sending state 0.00000 W with 2 decimals of accuracy
[12:32:56][D][sensor:127]: 'jbd-bms-uart discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart capacity remaining': Sending state 0.00000 Ah with 2 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart nominal capacity': Sending state 100.00000 Ah with 2 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart charging cycles': Sending state 0.00000  with 0 decimals of accuracy
[12:32:57][D][jbd_bms:266]:   Date of manufacture: 2022.2.16
[12:32:57][D][sensor:127]: 'jbd-bms-uart balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[12:32:57][D][text_sensor:067]: 'jbd-bms-uart errors': Sending state ''
[12:32:57][D][sensor:127]: 'jbd-bms-uart software version': Sending state 2.00000  with 1 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart operation status bitmask': Sending state 1.00000  with 0 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart battery strings': Sending state 16.00000  with 0 decimals of accuracy
[12:32:57][D][uart_debug:114]: <<< DD:03:00:17:00:00:00:00:00:00:27:10:00:00:2C:50:00:00:00:00:00:00:20:00:01:10:00:FF:05:77
[12:32:57][D][uart_debug:114]: >>> DD:A5:04:00:FF:FC:77
[12:32:57][D][sensor:127]: 'jbd-bms-uart cell voltage 1': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart cell voltage 2': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart cell voltage 3': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart cell voltage 4': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:57][D][sensor:127]: 'jbd-bms-uart cell voltage 5': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:58][D][sensor:127]: 'jbd-bms-uart cell voltage 6': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:58][D][sensor:127]: 'jbd-bms-uart cell voltage 7': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:58][D][sensor:127]: 'jbd-bms-uart cell voltage 8': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:58][D][sensor:127]: 'jbd-bms-uart cell voltage 9': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:58][D][sensor:127]: 'jbd-bms-uart cell voltage 10': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:58][D][sensor:127]: 'jbd-bms-uart cell voltage 11': Sending state 3.60000 V with 3 decimals of accuracy
[12:32:58][D][sensor:127]: 'jbd-bms-uart cell voltage 12': Sending state 3.60000 V with 3 decimals of accuracy
[... 28 lines cut ...]
[12:32:59][D][sensor:127]: 'jbd-bms-uart operation status bitmask': Sending state 1.00000  with 0 decimals of accuracy
[12:32:59][D][sensor:127]: 'jbd-bms-uart battery strings': Sending state 16.00000  with 0 decimals of accuracy
[12:32:59][D][uart_debug:114]: <<< DD:03:00:17:00:00:00:00:00:00:27:10:00:00:2C:50:00:00:00:00:00:00:20:00:01:10:00:FF:05:77
[12:32:59][D][uart_debug:114]: >>> DD:A5:04:00:FF:FC:77
[12:32:59][D][uart_debug:114]: <<< DD
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
[12:33:00][W][jbd_bms:124]: Invalid header
```

syssi, 2023-03-09:
That's for your feedback! How did you fix your temperature sensors?

syssi, 2023-03-09:
I've extended the list of supported devices! Feel free to create additional issues if you miss a feature or something doesn't work as expected.

closed 2023-03-09 by syssi