syssi/esphome-jbd-bms #49: automation problem ?

TheNexter, 2023-07-17:
Hello guy,

Your project is awsome, thankkk you. (sorry for my english)
I have setup an esp32 with home assistant and automation work great in home assistant but i do need to automate the charging stop at 80% for my office then i am trying to do it on the esp32 but it's not working, i disable the charge, i flash my esp32, home assistant have full control over my bms without any problem but after i plug the esp32 the charge need to start because my scooter is at 51%, do you have an idea ? 😞

Latest version is install 👍 

my yml automation :
```yaml
switch:
  - platform: ble_client
    ble_client_id: client0
    name: "${name} bluetooth connection"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    charging:
      name: "${name} charging"
      id: chargingtest
    discharging:
      name: "${name} discharging"

sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    state_of_charge:
      name: "${name} battery pourcentage"
      on_value_range:
        - above: 80
          then:
            - logger.log: "********* CHARGING OFF"
            - switch.turn_off: chargingtest
        - below: 80
          then:
            - logger.log: "********* CHARGING ON"
            - switch.turn_on: chargingtest
```
my full yml config :
```yaml
substitutions:
  name: battery-scooter
  device_description: "Monitor and control JBD-BMS via BLE"
  external_components_source: github://syssi/esphome-jbd-bms@main
  mac_address: # MAC ADDRESS OF MY BMS

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jbd-bms"
    version: 1.5.0

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  password: !secret esp_scooter_ota

api:
  encryption:
    key: !secret esp_scooter_api

logger:
  level: DEBUG

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("xiaoxiang", 0) == 0) {
            ESP_LOGI("ble_adv", "New JBD-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - id: client0
    mac_address: ${mac_address}

jbd_bms_ble:
  - id: bms0
    ble_client_id: client0
    update_interval: 2s

binary_sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    balancing:
      name: "${name} balancing"
    online_status:
      name: "${name} online status"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

switch:
  - platform: ble_client
    ble_client_id: client0
    name: "${name} bluetooth connection"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    charging:
      name: "${name} charging"
      id: chargingtest
    discharging:
      name: "${name} discharging"

sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    battery_strings:
      name: "${name} battery cells"
    charging_power:
      name: "${name} charging power"
    state_of_charge:
      name: "${name} battery pourcentage"
      on_value_range:
        - above: 80
          then:
            - logger.log: "********* CHARGING OFF"
            - switch.turn_off: chargingtest
        - below: 80
          then:
            - logger.log: "********* CHARGING ON"
            - switch.turn_on: chargingtest
    nominal_capacity:
      name: "${name} nominal capacity"
    charging_cycles:
      name: "${name} charging cycles"
    battery_cycle_capacity:
      name: "${name} battery cycle capacity"
    total_voltage:
      name: "${name} total voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
```

syssi, 2023-07-19:
Your automation looks good to me. I guess the `on_value_range` action triggers only if a defined value is crossed (79 -> 80). The transition from `NaN` to `80` won't be enough to turn off charging probably.

Please create a template `number` entity as fake state of charge value. Move the automation to this template and play around with the values of the number. Does your automation trigger as expected?

TheNexter, 2023-07-19:
> Your automation looks good to me. I guess the `on_value_range` action triggers only if a defined value is crossed (79 -> 80). The transition from `NaN` to `80` won't be enough to turn off charging [...]

Thank you, men, your message help me allot to understand the problem i have, i am gonna finish the perfect automation to manage battery charge and make a pull request for anyone who want to limit charge of there battery 😁 (and thx GPT4 too)

 ‼️  For anyone who use : 🛴🚲
If you use ebrake on your scooter, do not forget to enable charge before going on your scooter, without charge switch up on your BMS you cannot use ebrake to brake.
If you leave in house, you can have another esp32 that enable charge again when your scooter is close to go out of your garden.

TheNexter, 2023-07-19:
#50 Done, if you prefer in other way say me

syssi, 2023-07-19:
I left some comments. Could you improve your PR?

TheNexter, 2023-07-19:
> I left some comments. Could you improve your PR?

reply on PR

syssi, 2023-07-19:
Do you use a DIY battery pack for your scooter or was your scooter battery shipped with a JBD BMS?

TheNexter, 2023-07-19:
> Do you use a DIY battery pack for your scooter or was your scooter battery shipped with a JBD BMS?

DIY 😉

PR #50 (2023-07-20): Add YAML snippet how to preserve your battery pack
readme updated to include an explanation of automation to preserve battery life

Closes: #49