syssi/esphome-jbd-bms #52: Multiple BMS via BLE

Magsybaby, 2023-07-21:
Is is possible to monitor multiple BMS via BLE?

I've tried to add multiple by incrementing the IDs for the devices but not having much luck...wondering if it is me being dumb.

Thanks

syssi, 2023-07-21:
How many devices do you try to monitor? Could you provide your YAML? I will have a brief look. I guess you missed to change the sensor `name`s.

closed 2023-07-21 by syssi

avense, 2023-08-14:
it works very badly, that is, the sensors are interspersed between the bms0 and the bms1.
The total voltage sensor, for example, is interspersed between the bms0 and the bms1, that is, if I put this sensor on the bms0, the value of the bms1 appears as well.
Did I make myself understood?

syssi, 2023-08-14:
This shouldn't happen. Could you provide your YAML?

avense, 2023-08-14:
of course.

substitutions:
  name: baterias-lifepo4
  bms0: "${name} bms0"
  bms1: "${name} bms1"  
  device_description: "Monitor and control a Xiaoxiang Battery Management System (JBD-BMS) via BLE"
  external_components_source: github://syssi/esphome-jbd-bms@main
  bms0_mac_address: xxxxxxxxxxxxxxxx
  bms1_mac_address: xxxxxxxxxxx

esphome:
  name: ${name}
  platformio_options:
    platform_packages:
     - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
  comment: ${device_description}
  project:
    name: "syssi.esphome-jbd-bms"
    version: 1.5.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf 
external_components:
  - source: ${external_components_source} 
    refresh: 0s

wifi:
  ssid: xxxxxxx
  password: xxxxxxxxxxx
  manual_ip:
    static_ip: 192.168.0.113
    gateway: 192.168.0.1
    subnet: 255.255.255.0

ota:

logger:

#web_server:
# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: xxxxxxxxxxxx
  username: xxxxxxxxxxx
  password: xxxxxxxxxxx
  id: mqtt_client

#api:
#  password: 197842hdgp

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("xiaoxiang", 0) == 0) {
            ESP_LOGI("ble_adv", "New JBD-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
[... 162 lines cut ...]
    device_model:
      name: "${bms1} device model"

switch:
  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"
  
  - platform: ble_client
    ble_client_id: client1
    name: "${bms1} enable bluetooth connection"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    charging:
      name: "${bms0} charging"
    discharging:
      name: "${bms0} discharging"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms1
    charging:
      name: "${bms1} charging"
    discharging:
      name: "${bms1} discharging"

syssi, 2023-08-14:
Your YAML looks good if you use two different MAC addresses to connect to your devices.

As next step a ESPHome log would be nice. It should contain the ESPHome version (it's important to use the latest release), the BLE connection setup of both devices and some sensor readings. I hope this will help to identify the bug.

avense, 2023-08-14:
what logger level use?

syssi, 2023-08-14:
We should start with `DEBUG`.

syssi, 2023-08-14:
Which ESPHome version do you use?

avense, 2023-08-14:
[v2022.9.4](https://www.esphome.io/)

Magsybaby, 2023-08-14:
Apologies for not replying to this thread. The example YAML helped me, thank you so much for that. I have been monitoring two BMS for several weeks without issue.

I have used ESPHome 2023.7+

`substitutions:
  name: iot-bms-garage
  device_description: "Monitor and control a Xiaoxiang Battery Management System (JBD-BMS) via BLE"
  external_components_source: github://syssi/esphome-jbd-bms@main
  bms0: "30Q2P"
  bms1: "30Q4P"
  bms0_mac_address: A4:C1:38:CA:11:4F #30Q2P
  bms1_mac_address: A4:C1:38:2C:13:14 #30Q4P

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jbd-bms"
    version: 1.5.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  safe_mode: true
  password: !secret ota_password

logger:
  level: DEBUG
  logs:
    wifi: WARN

api:
  reboot_timeout: 0s
  encryption:
    key: "qNLKoxrZ5i+aFepfKas7bnsFiNCmu8kIaz/4WJ0Z2hk="

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

ble_client:
  - id: client0
    mac_address: ${bms0_mac_address}
  - id: client1
    mac_address: ${bms1_mac_address}

jbd_bms_ble:
  - id: bms0
    ble_client_id: client0
    update_interval: 300s
  - id: bms1
    ble_client_id: client1
    update_interval: 300s

binary_sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    balancing:
[... 187 lines cut ...]
      name: "${bms1} operation status"
    device_model:
      name: "${bms1} device model"

switch:
  - platform: ble_client
    ble_client_id: client0
    name: "${bms0} enable bluetooth connection"
  - platform: ble_client
    ble_client_id: client1
    name: "${bms1} enable bluetooth connection"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    charging:
      name: "${bms0} charging"
    discharging:
      name: "${bms0} discharging"
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms1
    charging:
      name: "${bms1} charging"
    discharging:
      name: "${bms1} discharging"
`

syssi, 2023-08-14:
@avense Please update ESPHome. There was a core bug in the past which is responsible for your issue.

@Magsybaby thanks for your feedback! :-)

avense, 2023-08-14:
@syssi same issue esphome last realease

log:

INFO ESPHome 2023.7.1
INFO Reading configuration /config/esphome/baterias-lifepo4.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/3500/head
INFO Starting log output from baterias-lifepo4/debug
INFO Connected to MQTT broker!
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 battery strings': Sending state 4.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 temperature': Sending state 26.30000 °C with 1 decimals of accuracy
[00:41:32][I][jbd_bms_ble:260]: Hardware info frame (34 bytes) received
[00:41:32][D][jbd_bms_ble:263]:   Device model: DP04S007L4S100A
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 total voltage': Sending state 13.22000 V with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 current': Sending state -10.16000 A with 1 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 power': Sending state -134.31519 W with 1 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 charging power': Sending state 0.00000 W with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 discharging power': Sending state 134.31519 W with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 capacity remaining': Sending state 176.86000 Ah with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 nominal capacity': Sending state 230.00000 Ah with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 charging cycles': Sending state 1.00000  with 0 decimals of accuracy
[00:41:32][D][jbd_bms_ble:290]:   Date of manufacture: 2022.9.12
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[00:41:32][D][text_sensor:067]: 'baterias-lifepo4 bms1 errors': Sending state ''
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 software version': Sending state 6.00000  with 1 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 state of charge': Sending state 77.00000 % with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 battery strings': Sending state 4.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms1 temperature': Sending state 26.30000 °C with 1 decimals of accuracy
[00:41:32][I][jbd_bms_ble:260]: Hardware info frame (34 bytes) received
[00:41:32][D][jbd_bms_ble:263]:   Device model: DP04S007L4S100A
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 total voltage': Sending state 13.19000 V with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 current': Sending state -10.25000 A with 1 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 power': Sending state -135.19749 W with 1 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 charging power': Sending state 0.00000 W with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 discharging power': Sending state 135.19749 W with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 capacity remaining': Sending state 175.76999 Ah with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 nominal capacity': Sending state 230.00000 Ah with 2 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 charging cycles': Sending state 25.00000  with 0 decimals of accuracy
[00:41:32][D][jbd_bms_ble:290]:   Date of manufacture: 2023.3.16
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[00:41:32][D][text_sensor:067]: 'baterias-lifepo4 bms0 errors': Sending state ''
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 software version': Sending state 2.40000  with 1 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 state of charge': Sending state 76.00000 % with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 battery strings': Sending state 4.00000  with 0 decimals of accuracy
[00:41:32][D][sensor:127]: 'baterias-lifepo4 bms0 temperature': Sending state 26.20000 °C with 1 decimals of accuracy
[00:41:32][I][jbd_bms_ble:260]: Hardware info frame (34 bytes) received
[00:41:32][D][jbd_bms_ble:263]:   Device model: DP04S007L4S100A

syssi, 2023-08-15:
Could you provide your YAML another time? The log book doesn't match with the provided YAML above.

syssi, 2023-08-15:
Btw the log doesn't show any issues:

```
sebastian@wolowitz:/tmp$ cat b.txt 
'baterias-lifepo4 bms0 balancer status bitmask': Sending state 0.00000 with 0 decimals of accuracy
'baterias-lifepo4 bms1 balancer status bitmask': Sending state 0.00000 with 0 decimals of accuracy

'baterias-lifepo4 bms0 battery strings': Sending state 4.00000 with 0 decimals of accuracy
'baterias-lifepo4 bms1 battery strings': Sending state 4.00000 with 0 decimals of accuracy

'baterias-lifepo4 bms0 battery strings': Sending state 4.00000 with 0 decimals of accuracy

'baterias-lifepo4 bms0 capacity remaining': Sending state 175.76999 Ah with 2 decimals of accuracy
'baterias-lifepo4 bms1 capacity remaining': Sending state 176.86000 Ah with 2 decimals of accuracy

'baterias-lifepo4 bms0 charging cycles': Sending state 25.00000 with 0 decimals of accuracy
'baterias-lifepo4 bms1 charging cycles': Sending state 1.00000 with 0 decimals of accuracy

'baterias-lifepo4 bms0 charging power': Sending state 0.00000 W with 2 decimals of accuracy
'baterias-lifepo4 bms1 charging power': Sending state 0.00000 W with 2 decimals of accuracy

'baterias-lifepo4 bms0 current': Sending state -10.25000 A with 1 decimals of accuracy
'baterias-lifepo4 bms1 current': Sending state -10.16000 A with 1 decimals of accuracy

'baterias-lifepo4 bms0 discharging power': Sending state 135.19749 W with 2 decimals of accuracy
'baterias-lifepo4 bms1 discharging power': Sending state 134.31519 W with 2 decimals of accuracy

'baterias-lifepo4 bms0 errors bitmask': Sending state 0.00000 with 0 decimals of accuracy
'baterias-lifepo4 bms1 errors bitmask': Sending state 0.00000 with 0 decimals of accuracy

'baterias-lifepo4 bms0 errors': Sending state ''
'baterias-lifepo4 bms1 errors': Sending state ''

'baterias-lifepo4 bms0 nominal capacity': Sending state 230.00000 Ah with 2 decimals of accuracy
'baterias-lifepo4 bms1 nominal capacity': Sending state 230.00000 Ah with 2 decimals of accuracy

'baterias-lifepo4 bms0 operation status bitmask': Sending state 3.00000 with 0 decimals of accuracy
'baterias-lifepo4 bms1 operation status bitmask': Sending state 3.00000 with 0 decimals of accuracy

'baterias-lifepo4 bms0 power': Sending state -135.19749 W with 1 decimals of accuracy
'baterias-lifepo4 bms1 power': Sending state -134.31519 W with 1 decimals of accuracy

'baterias-lifepo4 bms0 software version': Sending state 2.40000 with 1 decimals of accuracy
'baterias-lifepo4 bms1 software version': Sending state 6.00000 with 1 decimals of accuracy

'baterias-lifepo4 bms0 state of charge': Sending state 76.00000 % with 0 decimals of accuracy
'baterias-lifepo4 bms1 state of charge': Sending state 77.00000 % with 0 decimals of accuracy

'baterias-lifepo4 bms0 temperature': Sending state 26.20000 °C with 1 decimals of accuracy
'baterias-lifepo4 bms1 temperature': Sending state 26.30000 °C with 1 decimals of accuracy

'baterias-lifepo4 bms0 total voltage': Sending state 13.19000 V with 2 decimals of accuracy
'baterias-lifepo4 bms1 total voltage': Sending state 13.22000 V with 2 decimals of accuracy

Date of manufacture: 2022.9.12
Date of manufacture: 2023.3.16
```

avense, 2023-08-15:
@syssi But in this sensor for example "sensor.baterias_lifepo4_bms0_total_voltage" why does the value of bms1 appear interspersed with bms0?

syssi, 2023-08-15:
Do you see the mix up of the sensor readings at the log too? Could you provide a longer capture with multiple `total voltage` readings of both devices?

avense, 2023-08-15:
do you want to see the log?

syssi, 2023-08-16:
Do you use this package on purpose?

```
esphome:
  name: ${name}
  platformio_options:
    platform_packages:
     - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
```

If not please remove these lines from your YAML:

```
  platformio_options:
    platform_packages:
     - toolchain-riscv3
```

Using this pull request isn't required anymore. The new `web_server_ idf` component is merged and part of the latest release. Please remove these lines too:

```
external_components:
  - source: ${external_components_source} 
  - source: github://pr#3500
    components:
      # list all components modified by this Pull Request here
      - web_server
      - web_server_idf
      - web_server_base
      - captive_portal
    refresh: 0s
```

Could you open the web interface and provide another short clip / recording of the `total voltage` sensor here? Thanks in advance!

avense, 2023-08-16:
external_components:
  - source: ${external_components_source} 
  - source: github://pr#3500
    components:
      # list all components modified by this Pull Request here
      - web_server
      - web_server_idf
      - web_server_base
      - captive_portal
    refresh: 0s

If you remove this option, you get this error.:

INFO ESPHome 2023.7.1
INFO Reading configuration /config/esphome/baterias-lifepo4.yaml...
Failed config

jbd_bms_ble: [source /config/esphome/baterias-lifepo4.yaml:73]
  
  Component not found: jbd_bms_ble.
  - id: bms0
    ble_client_id: client0
    update_interval: 5s
  - id: bms1
    ble_client_id: client1
    update_interval: 5s
binary_sensor.jbd_bms_ble: [source /config/esphome/baterias-lifepo4.yaml:82]
  
  Platform not found: 'binary_sensor.jbd_bms_ble'.
  platform: jbd_bms_ble
  jbd_bms_ble_id: bms0
  balancing: 
    name: baterias-lifepo4 bms0 balancing
  charging: 
    name: baterias-lifepo4 bms0 charging
  discharging: 
    name: baterias-lifepo4 bms0 discharging
  online_status: 
    name: baterias-lifepo4 bms0 online status
binary_sensor.jbd_bms_ble: [source /config/esphome/baterias-lifepo4.yaml:93]
  
  Platform not found: 'binary_sensor.jbd_bms_ble'.
  platform: jbd_bms_ble
  jbd_bms_ble_id: bms1
  balancing: 
    name: baterias-lifepo4 bms1 balancing
  charging: 
    name: baterias-lifepo4 bms1 charging
  discharging: 
    name: baterias-lifepo4 bms1 discharging
  online_status: 
    name: baterias-lifepo4 bms1 online status
sensor.jbd_bms_ble: [source /config/esphome/baterias-lifepo4.yaml:105]
  
  Platform not found: 'sensor.jbd_bms_ble'.
  platform: jbd_bms_ble
  jbd_bms_ble_id: bms0
  battery_strings: 
    name: baterias-lifepo4 bms0 battery strings
  current: 
    name: baterias-lifepo4 bms0 current
  power: 
    name: baterias-lifepo4 bms0 power
  charging_power: 
    name: baterias-lifepo4 bms0 charging power
  discharging_power: 
    name: baterias-lifepo4 bms0 discharging power
  state_of_charge: 
    name: baterias-lifepo4 bms0 state of charge
  nominal_capacity: 
[... 111 lines cut ...]
  jbd_bms_ble_id: bms1
  errors: 
    name: baterias-lifepo4 bms1 errors
  operation_status: 
    name: baterias-lifepo4 bms1 operation status
  device_model: 
    name: baterias-lifepo4 bms1 device model
switch.jbd_bms_ble: [source /config/esphome/baterias-lifepo4.yaml:245]
  
  Platform not found: 'switch.jbd_bms_ble'.
  platform: jbd_bms_ble
  jbd_bms_ble_id: bms0
  charging: 
    name: baterias-lifepo4 bms0 charging
  discharging: 
    name: baterias-lifepo4 bms0 discharging
switch.jbd_bms_ble: [source /config/esphome/baterias-lifepo4.yaml:252]
  
  Platform not found: 'switch.jbd_bms_ble'.
  platform: jbd_bms_ble
  jbd_bms_ble_id: bms1
  charging: 
    name: baterias-lifepo4 bms1 charging
  discharging: 
    name: baterias-lifepo4 bms1 discharging

syssi, 2023-08-16:
My fault. This lines are still required to load this external component / project:

```
external_components:
  - source: ${external_components_source
```

avense, 2023-08-16:
It's finally good.....
Thank you very much friend, and good job.
I'm going on my youtube channel to mention your work.
Thank you very much

commit 8c282672 (2023-07-21): Add monitor multiple devices via BLE example (Closes: #52) (#53)