syssi/esphome-jbd-bms #55: Force wait value before api / mqtt available

TheNexter, 2023-07-23:
Hi men, is there is a way to force the esp to wait some or all sensor / switch to be avalaible before making the api or the mqtt available ?
I ask this because in my HA, everytime i go deepsleep this appends
[image]

But when i disable it > working without any problem 😅

This could be so great when i leave with my scooter no error message on HA and when i am back to home, just update of value when the value is ready and not before or in some part like now 😅

SORRY for my english 🫠🫠🫠🫠

TheNexter, 2023-07-23:
My full config :

``` YAML
substitutions:
  name: scooter-battery
  bms0: "BMS0"
  bms1: "BMS1"
  device_description: "Monitor and control a Xiaoxiang Battery Management System (JBD-BMS) via BLE"
  external_components_source: github://syssi/esphome-jbd-bms@main
  bms0_mac_address: # BMS 0
  bms1_mac_address: # BMS 1

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jbd-bms"
    version: 1.5.0

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.61
    gateway: 192.168.0.254
    subnet: 255.255.255.0
  reboot_timeout: 0s

ota:
  password: !secret esp_scooter_ota

api:
  encryption:
    key: !secret esp_scooter_api
  reboot_timeout: 0s

deep_sleep:
  id: deep_sleep0
  run_duration: 2min
  sleep_duration: 10min

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("xiaoxiang", 0) == 0) {
            ESP_LOGI("ble_adv", "New JBD-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
  - mac_address: ${bms1_mac_address}
    id: client1

[... 190 lines cut ...]
    temperature_2:
      name: "${bms1} temperature 2"
    temperature_3:
      name: "${bms1} temperature 3"
    cell_voltage_1:
      name: "${bms1} cell voltage 1"
    cell_voltage_2:
      name: "${bms1} cell voltage 2"
    cell_voltage_3:
      name: "${bms1} cell voltage 3"
    cell_voltage_4:
      name: "${bms1} cell voltage 4"
    cell_voltage_5:
      name: "${bms1} cell voltage 5"
    cell_voltage_6:
      name: "${bms1} cell voltage 6"
    cell_voltage_7:
      name: "${bms1} cell voltage 7"
    cell_voltage_8:
      name: "${bms1} cell voltage 8"
    cell_voltage_9:
      name: "${bms1} cell voltage 9"
    cell_voltage_10:
      name: "${bms1} cell voltage 10"
```

syssi, 2023-07-25:
I must admit I don't understand the screenshot above. First of all: The entity names of the sensor at the YAML are a bit different to the entity names at the screenshot. Is the screenshot out of sync to the YAML configuration or did you rename the entities?

What happens to the measurements if your node goes into deep sleep? Do you see the last measurement or something like `unavailable` at the right column? If you reboot Home Assistant and the ESPHome node isn't available because of deep sleep the yellow triangle warnings does show up? Could you try to provide some more details about the issue?

TheNexter, 2023-07-25:
Hello, i just rename my sensor 😅

I have try to restart HA, same problem 🫠

> What happens to the measurements if your node goes into deep sleep

The data stay here without problem when deep sleep is on but few deep sleep loop the esp start to send data to home assistant without having the true data and then the esp send "unvailable" (see screenshot)

> Could you try to provide some more details about the issue?

It's simple : Is there is a way to NOT send data to HA until when esp is connected to a bms battery scooter and retrieve true data and not "unvailable" ?