syssi/esphome-jbd-bms #60: JBD-UP16S010-L16S-100A

mykolavorobiov, 2023-09-15:
Could you help with new bms
I see that esp host send 
[image]
And has no response
But on protocol description I see that I should send ```DD 00 A5 03 00 FF 58 77```
[image]

Is it possible at all?

Esp32 connected via UART (lcd port)

syssi, 2023-10-31:
Are you still willing to test your protocol flavor? I would provide a first attempt.

mykolavorobiov, 2023-11-02:
Sorry for my late, I think that it was a problem with my hardware, coz I'm using MAX485 UART-RS485 without automatic flow control pin.
I [forked your repo](https://github.com/mykolavorobiov/esphome-jbd-bms-with-flow-pin) and make several changes to add flow pin.
It works correct for on my side now.

closed 2023-11-02 by mykolavorobiov