syssi/esphome-jbd-bms #62: Is there a way to just get the data from serial?

TjadenWright, 2023-10-30:
I want to just get the data from the battery through bluetooth to the esp32. Then the data in the esp32 will be sent to my computer through serial. Is there a way to disable WIFI and MQTT so that it only shows the data from bluetooth?

syssi, 2023-10-30:
Do you want to parse the log messages or do you want to implement a serial protocol?

syssi, 2023-10-30:
If you don't want to use MQTT just remove the `mqtt` section. Same applies to `wifi`. The components aren't required. I would disable the `logger` and write a message per sensor value to a dedicated `uart`.

TjadenWright, 2023-10-30:
This works perfectly thank you! All I need was to get rid of it looking for wifi and mqtt all the time.

closed 2023-10-30 by syssi