syssi/esphome-jbd-bms #67: Invalid header

taHC81, 2023-11-27:
Hello, I'm trying to use ESP32 with LAN8720, it's working almost fine with one issue. jbd-bms component parses the Hardware info properly, but Cell info fails with multiple **Invalid header** error. But uart_debug returns valid response.

```
[09:05:31][D][uart_debug:114]: >>> DD:A5:03:00:FF:FD:77
[09:05:31][I][jbd_bms:249]: Hardware info frame (38 bytes) received
[09:05:31][D][jbd_bms:252]:   Device model: JBD-AP21S001-L21S-200A-B
[09:05:31][D][sensor:094]: 'jbd-bms total voltage': Sending state 51.87000 V with 2 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms current': Sending state 1.23000 A with 1 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms power': Sending state 63.80010 W with 1 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms charging power': Sending state 63.80010 W with 2 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms capacity remaining': Sending state 85.89000 Ah with 2 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms nominal capacity': Sending state 30.00000 Ah with 2 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms charging cycles': Sending state 143.00000  with 0 decimals of accuracy
[09:05:31][D][jbd_bms:279]:   Date of manufacture: 2022.12.19
[09:05:31][D][sensor:094]: 'jbd-bms balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[09:05:31][D][text_sensor:064]: 'jbd-bms errors': Sending state ''
[09:05:31][D][sensor:094]: 'jbd-bms software version': Sending state 4.40000  with 1 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms state of charge': Sending state 31.00000 % with 0 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms battery strings': Sending state 16.00000  with 0 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms temperature 1': Sending state 15.00000 °C with 1 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms temperature 2': Sending state 16.10000 °C with 1 decimals of accuracy
[09:05:31][D][sensor:094]: 'jbd-bms temperature 3': Sending state 16.40000 °C with 1 decimals of accuracy
[09:05:31][D][uart_debug:114]: <<< DD:03:00:26:14:43:00:7B:21:8D:0B:B8:00:8F:2D:93:00:00:00:00:00:00:44:1F:03:10:03:0B:41:0B:4C:0B:4F:00:00:00:6D:60:21:8D:00:00:F9:57:77
[09:05:31][D][uart_debug:114]: >>> DD:A5:04:00:FF:FC:77
[09:05:31][W][component:214]: Component jbd_bms took a long time for an operation (0.22 s).
[09:05:31][W][component:215]: Components should block for at most 20-30ms.
[09:05:31][W][jbd_bms:131]: Invalid header: 0x9E
[09:05:31][W][jbd_bms:131]: Invalid header: 0x0C
[09:05:31][W][jbd_bms:131]: Invalid header: 0xA4
[09:05:31][W][jbd_bms:131]: Invalid header: 0x0C
[09:05:31][W][jbd_bms:131]: Invalid header: 0xAD
[09:05:31][W][jbd_bms:131]: Invalid header: 0x0C
[09:05:31][W][jbd_bms:131]: Invalid header: 0xAF
[09:05:31][W][jbd_bms:131]: Invalid header: 0x0C
[09:05:31][W][jbd_bms:131]: Invalid header: 0xB1
[09:05:31][W][jbd_bms:131]: Invalid header: 0x0C
[09:05:31][W][jbd_bms:131]: Invalid header: 0xA8
[09:05:31][W][jbd_bms:131]: Invalid header: 0x0C
[09:05:31][W][jbd_bms:131]: Invalid header: 0xAB
[09:05:31][W][jbd_bms:131]: Invalid header: 0xF4
[09:05:31][W][jbd_bms:131]: Invalid header: 0x7D
[09:05:31][W][jbd_bms:131]: Invalid header: 0x77
[09:05:31][W][component:214]: Component jbd_bms took a long time for an operation (0.06 s).
[09:05:31][W][component:215]: Components should block for at most 20-30ms.
[09:05:31][D][uart_debug:114]: <<< DD:04:00:20:0C:AE:0C:AD:0C:AD:0C:B1:0C:B0:0C:AF:0C:A4:0C:9F:0C:A6:0C:9E:0C:A4:0C:AD:0C:AF:0C:B1:0C:A8:0C:AB:F4:7D:77
```

taHC81, 2023-11-27:
Even stranger is when I trigger the Cell info request via recently created button within ESPhome - it works. And uart_debug reports the same TX message (DD:A5:04:00:FF:FC:77). I'm using GPIO14 for TX and GPIO13 for RX, device is ESP-WROOM-32.
```

button:
  - platform: template
    name: "Request cell info"
    on_press:
      - uart.write: [0xDD, 0xA5, 0x04, 0x00, 0xFF, 0xFC, 0x77]
```

Log:

```
[09:05:25][D][button:010]: 'Request cell info' Pressed.
[09:05:25][D][uart_debug:114]: >>> DD:A5:04:00:FF:FC:77
[09:05:25][I][jbd_bms:198]: Cell info frame (32 bytes) received
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 01': Sending state 3.24600 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 02': Sending state 3.24500 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 03': Sending state 3.24500 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 04': Sending state 3.24700 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 05': Sending state 3.24800 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 06': Sending state 3.24700 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 07': Sending state 3.23600 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 08': Sending state 3.23300 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 09': Sending state 3.23800 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 10': Sending state 3.23000 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 11': Sending state 3.23600 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 12': Sending state 3.24500 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 13': Sending state 3.24700 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 14': Sending state 3.24900 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 15': Sending state 3.24000 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms cell voltage 16': Sending state 3.24500 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms min cell voltage': Sending state 3.23000 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms max cell voltage': Sending state 3.24900 V with 3 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms max voltage cell': Sending state 14.00000  with 0 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms min voltage cell': Sending state 10.00000  with 0 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms delta cell voltage': Sending state 0.01900 V with 4 decimals of accuracy
[09:05:25][D][sensor:094]: 'jbd-bms average cell voltage': Sending state 3.24231 V with 4 decimals of accuracy
[09:05:25][W][component:214]: Component jbd_bms took a long time for an operation (0.21 s).
[09:05:25][W][component:215]: Components should block for at most 20-30ms.
[09:05:25][D][uart_debug:114]: <<< DD:04:00:20:0C:AE:0C:AD:0C:AD:0C:AF:0C:B0:0C:AF:0C:A4:0C:A1:0C:A6:0C:9E:0C:A4:0C:AD:0C:AF:0C:B1:0C:A8:0C:AD:F4:7B:77
```

syssi, 2023-11-27:
Could you provide some more logs including the raw traffic (like above but a few minutes of communication)? May be the size of the device info frame has changed and confuses the receiver log / frame parser now.

syssi, 2023-11-27:
Could you provide some YAML configuration too? I would like to make sure the `dummy_receiver` of the `uart.debug` feature isn't enabled.

taHC81, 2023-11-27:
Here's the YAML - https://pastebin.com/zLDbVMJj

syssi, 2023-11-27:
The YAML looks good!

taHC81, 2023-11-27:
Here's the log - https://pastebin.com/jRYF8MZd

syssi, 2023-11-27:
How long is the wire between the JBD and ESP?

taHC81, 2023-11-27:
> How long is the wire between the JBD and ESP?

Initially running with ~1 meter cable, 2nd run with original cable coming with JBD BMS = 20cm. Still the same.

syssi, 2023-11-27:
Is there is difference if there is load (discharging) vs charging?

taHC81, 2023-11-27:
During the day was battery slightly charged (200-700W), but inverter still in bypass mode, thus no EMI radiation. Let's wait for the WT32-ETH01 module to try with. Current one is a ghetto development :)

taHC81, 2023-11-28:
Good news, I've switched from arduino to esp-idf framework and it's much more reliable. I'd say, it's rock stable, but need longer time to prove that. Here's the log - https://pastebin.com/3iA4Rgtz
```

esp32:
  board: wemos_d1_mini32
  framework:
    # type: arduino
    # version: latest
    type: esp-idf
    version: recommended
```

syssi, 2023-11-28:
This looks pretty good now! Please monitor the behavior for a few days under different conditions. It's still possible the situation gets worse as soon there is some load.

See https://github.com/syssi/esphome-jbd-bms/issues/4

dougle03, 2023-12-05:
> Good news, I've switched from **arduino** to **esp-idf** framework and it's much more reliable. I'd say, it's rock stable, but need longer time to prove that. Here's the log - [...]

Hi, did it remain stable?
Thanks

taHC81, 2023-12-10:
Hi there, still haven't found enough time to switch to this solution, sorry. But I'll do that soon ;)

taHC81, 2023-12-18:
Hi all, today is the day when I switched to ethernet version, looks good for last hour, but will report later this week. Treat it as x-mas gift :)

taHC81, 2023-12-19:
Nicely running for last 24 hours without any hiccup, there was 1.8 kW charging peak yesterday and some moderate load (1.4 kW at most). Really happy how it is working!
[image]

syssi, 2023-12-19:
Good job! Let's close this issue.

closed 2023-12-19 by syssi