syssi/esphome-jbd-bms #69: Add reset ESP button

hddmax, 2024-01-02:
can you add the reset buton ?

i have tried it to add id but it doesnt work.

button:
  - platform: restart
    name: "ESP_JBD_reset"

syssi, 2024-01-22:
Please try a clean build. It should just work!

closed 2024-01-22 by syssi

hddmax, 2024-01-28:
i tried it but it doesnt work ? hmmm

Compiling .pioenvs/jbd-bms-ble/src/main.o
Linking .pioenvs/jbd-bms-ble/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/jbd-bms-ble/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection14button_commandERKNS0_20ButtonCommandRequestE+0x0): undefined reference to `esphome::button::Button::press()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/jbd-bms-ble/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::button_command(esphome::api::ButtonCommandRequest const&)':
/data/build/jbd-bms-ble/src/esphome/components/api/api_connection.cpp:795: undefined reference to `esphome::button::Button::press()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/jbd-bms-ble/src/main.o:(.literal._Z5setupv+0x760): undefined reference to `vtable for esphome::restart::RestartButton'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/jbd-bms-ble/src/main.o:(.literal._Z5setupv+0x764): undefined reference to `vtable for esphome::restart::RestartButton'
collect2: error: ld returned 1 exit status
*** [.pioenvs/jbd-bms-ble/firmware.elf] Error 1
========================= [FAILED] Took 79.24 seconds =========================

syssi, 2024-01-28:
Is this a clean build for sure? Please provide your complete YAML configuration. I would like to reproduce the build issue locally.

hddmax, 2024-01-28:
yes i did a clean yaml over HA,

here the full code:

`substitutions:
  name: jbd-bms-ble
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  bms2: "${name} bms2"
  external_components_source: github://syssi/esphome-jbd-bms@main
  bms0_mac_address: A4:C1:38:2A:95:DB
  bms1_mac_address: A4:C1:38:A2:EA:D8
  bms2_mac_address: A4:C1:38:E6:3F:18

esphome:
  name: ${name}
    
esp32:
  board: esp32dev
  framework:
    type: esp-idf
   
external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  password: "cf7319c6dd5079cad5a9243384ae83d3"

logger:
  level: DEBUG

api:

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("xiaoxiang", 0) == 0) {
            ESP_LOGI("ble_adv", "New JBD-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
  - mac_address: ${bms1_mac_address}
    id: client1
  - mac_address: ${bms2_mac_address}
    id: client2

jbd_bms_ble:
  - id: bms0
    ble_client_id: client0
    update_interval: 15s
  - id: bms1
    ble_client_id: client1
    update_interval: 15s
  - id: bms2
    ble_client_id: client2
    update_interval: 15s
[... 304 lines cut ...]
    jbd_bms_ble_id: bms0
    charging:
      name: "${bms0} charging"
    discharging:
      name: "${bms0} discharging"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms1
    charging:
      name: "${bms1} charging"
    discharging:
      name: "${bms1} discharging"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms2
    charging:
      name: "${bms2} charging"
    discharging:
      name: "${bms2} discharging"   

# button:
#  - platform: restart
#    name: "ESP_JBD_reset"      `

the reset is only for tests remarked

syssi, 2024-01-28:
Please make sure your ESPHome version is up-to-date (`2023.12.9`). The YAML builds fine here:

```
Linking .pioenvs/jbd-bms-ble/bootloader.elf
Building .pioenvs/jbd-bms-ble/bootloader.bin
Creating esp32 image...
Successfully created esp32 image.
Linking .pioenvs/jbd-bms-ble/firmware.elf
RAM:   [==        ]  16.1% (used 52796 bytes from 327680 bytes)
Flash: [=======   ]  65.6% (used 1203065 bytes from 1835008 bytes)
Building .pioenvs/jbd-bms-ble/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/jbd-bms-ble/firmware.bin"], [".pioenvs/jbd-bms-ble/firmware.elf"])
Wrote 0x1371f0 bytes to file /tmp/.esphome/build/jbd-bms-ble/.pioenvs/jbd-bms-ble/firmware-factory.bin, ready to flash to offset 0x0
==================================================================== [SUCCESS] Took 71.98 seconds ====================================================================
INFO Successfully compiled program.
```

hddmax, 2024-01-28:
now it works , i have no glue why :-)

[image]