syssi/esphome-jbd-bms #70: SocketAPIError

OLDDRUNK, 2024-01-03:
New install with faker config. 
Have restarted HA.
Not sure if this is my end or other settings?

```
INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/esp8266.yaml...
INFO Updating https://github.com/syssi/esphome-jbd-bms.git@main
INFO Starting log output from 192.168.1.152 using esphome API
WARNING Can't connect to ESPHome API for jbd-bms-uart @ 192.168.1.152: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.152', port=6053))]: [Errno 113] Connect call failed ('192.168.1.152', 6053) (SocketAPIError)
INFO Trying to connect to jbd-bms-uart @ 192.168.1.152 in the background
```
```
substitutions:
  name: jbd-bms-uart
  device_description: "Monitor and control a Xiaoxiang Battery Management System (JBD-BMS) via UART"
  external_components_source: github://syssi/esphome-jbd-bms@main
  tx_pin: GPIO4
  rx_pin: GPIO5
  rx_timeout: 150ms

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jbd-bms"
    version: 1.5.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.152
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  reboot_timeout: 0s

ota:
  password: !secret ota

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

api:
  encryption:
    key: !secret apikk
  reboot_timeout: 0s

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

jbd_bms:
  - id: bms0
    uart_id: uart_0
    update_interval: 2s
    rx_timeout: ${rx_timeout}
    enable_fake_traffic: true

binary_sensor:
  - platform: jbd_bms
    jbd_bms_id: bms0
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    online_status:
      name: "${name} online status"

sensor:
  - platform: jbd_bms
    jbd_bms_id: bms0
    battery_strings:
      name: "${name} battery strings"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    state_of_charge:
      name: "${name} state of charge"
    nominal_capacity:
      name: "${name} nominal capacity"
    charging_cycles:
      name: "${name} charging cycles"
    capacity_remaining:
      name: "${name} capacity remaining"
    battery_cycle_capacity:
      name: "${name} battery cycle capacity"
    total_voltage:
      name: "${name} total voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    temperature_1:
      name: "${name} temperature 1"
    temperature_2:
      name: "${name} temperature 2"
    temperature_3:
      name: "${name} temperature 3"
    temperature_4:
      name: "${name} temperature 4"
    temperature_5:
      name: "${name} temperature 5"
    temperature_6:
      name: "${name} temperature 6"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    cell_voltage_17:
      name: "${name} cell voltage 17"
    cell_voltage_18:
      name: "${name} cell voltage 18"
    cell_voltage_19:
      name: "${name} cell voltage 19"
    cell_voltage_20:
      name: "${name} cell voltage 20"
    cell_voltage_21:
      name: "${name} cell voltage 21"
    cell_voltage_22:
      name: "${name} cell voltage 22"
    cell_voltage_23:
      name: "${name} cell voltage 23"
    cell_voltage_24:
      name: "${name} cell voltage 24"
    cell_voltage_25:
      name: "${name} cell voltage 25"
    cell_voltage_26:
      name: "${name} cell voltage 26"
    cell_voltage_27:
      name: "${name} cell voltage 27"
    cell_voltage_28:
      name: "${name} cell voltage 28"
    cell_voltage_29:
      name: "${name} cell voltage 29"
    cell_voltage_30:
      name: "${name} cell voltage 30"
    cell_voltage_31:
      name: "${name} cell voltage 31"
    cell_voltage_32:
      name: "${name} cell voltage 32"
    operation_status_bitmask:
      name: "${name} operation status bitmask"
    errors_bitmask:
      name: "${name} errors bitmask"
    balancer_status_bitmask:
      name: "${name} balancer status bitmask"
    software_version:
      name: "${name} software version"

text_sensor:
  - platform: jbd_bms
    jbd_bms_id: bms0
    errors:
      name: "${name} errors"
    operation_status:
      name: "${name} operation status"
    device_model:
      name: "${name} device model"

switch:
  - platform: jbd_bms
    jbd_bms_id: bms0
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

```

syssi, 2024-01-03:
Could you remove the MQTT section just for testing? What kind of MQTT broker do you use?

OLDDRUNK, 2024-01-03:
I use mosquitto. Will remove mqtt and report

syssi, 2024-01-03:
Another idea: The `logger` writes all log message to UART0 (GPIO1/GPIO3). If you attach the Wemos D1 Mini to a computer using an USB cable you should see the boot-loop and probably the causing exception.

OLDDRUNK, 2024-01-03:
Okay works perfectly with mqtt removed. What could be going on there?

Unbelievably good build! This is genuinely life changing!

syssi, 2024-01-03:
Alright. Let's add the `mqtt` section again and remove all sensors except a single one like `total voltage`. Build and flash the ESP another time. If it's a problem because of the limited resources of the ESP it could work this time. If the ESP is still crashing I expect too much traffic on your MQTT broker which gets pushed to the ESP on login.

syssi, 2024-01-22:
Any updates? I would like to close the issue.

closed 2024-01-24 by syssi