syssi/esphome-jbd-bms #78: RS485 <-> TTL Adapter

preussal, 2024-03-15:
Is it possible to put an RS485 <-> TTL adapter between the ESP32 TX / RX and the BMS A / B in between? 

My BMS has the RS485 A/B Port and the UART Port where the Bluetooth modules are on it.
I've already tried it with Bluetooth but I don't get any usable data. Everything takes so long.
When I disconnect the Bluetooth Adapter and connect the TX/RX Pins to the ESP32 I get any header error.

But the JBD UART - USB Black Box with the PC works then i connect them to the UART Port

syssi, 2024-03-20:
I don't know any details about the mentioned port. The RS485 port of my JBD BMS isn't equipped. Could you try to ask your seller about some details?

syssi, 2024-03-20:
> This protocol supports the universal protocol of the RS485 / RS232 / UART interface of the JBD smart bms, same with the host computer protocol. The baud rate is 9600BPS or other custom rates.

It could work out of the box if we are lucky! Just attach a RS485 converter to your ESP and give it a try!

preussal, 2024-04-03:
Sorry for the late reply. So with an RS465 to TTL board, the connection works.
[image]
It's not the most beautiful cabling yet, but that is still adjusted.

closed 2024-04-12 by preussal