syssi/esphome-jbd-bms #81: Connectionproblem Liontron BMS

jassi0001, 2024-07-15:
I have a problem connection my liontron bms. BMS Firmware-Version is 2.5.
When i turn off the bluetooth in the website of the esphome interface and reacitvate it, is get values for one time. In the same session the bms is shown to ha as bms errors and going to offline.
I have esphome 2024.06.04 compiled with idf.
In the interface i get [D]	[esp32_ble_client:110]	[0] [70:3E:97:xx:xx:xx] ESP_GATTC_WRITE_CHAR_EVT
The connection is still blocked than, because i can not find the bms with the mobile phone app. When i disable bluetooth in esphome, the bms can immediately be found with the phone app
When connection is not working …. Hopefully you can help here.

syssi, 2024-07-15:
Do you have a Android device and are you able to prepare a btsnoop capture of the BLE traffic? I would like to compare the traffic of the Android App with the implementation here. There is probably a difference (in the command set) which confuses the BMS.

jassi0001, 2024-07-15:
Not at the moment. I only have iphone available.
i just installed 2024.6.6 and activated some debug and verbose logging…
i get this at the moment, without response from the bms it seems….
	
Send command: DD.A5.03.00.FF.FD.77 (7)
22:57:12	[V]	[esp32_ble_client:121]	
[0] [70:3E:97:D1:C1:8C] gattc_event_handler: event=4 gattc_if=3
22:57:12	[D]	[esp32_ble_client:110]	
[0] [70:3E:97:D1:C1:8C] ESP_GATTC_WRITE_CHAR_EVT

syssi, 2024-07-15:
Could you provide a download link to the Android app of your BMS?

syssi, 2024-07-15:
The log messages aren't helpful without any context.

jassi0001, 2024-07-15:
This one: https://play.google.com/store/apps/details?id=com.liontron.liontronmulti&hl=de&pli=1

syssi, 2024-07-16:
Die App schickt die folgenden Befehle/Frames periodisch an die App:

```
DD A5 03 00 FF FD 77
DD A5 04 00 FF FC 77
```

syssi, 2024-07-16:
Expected response: The start of frame is `DD` and the end of frame `77`.

The app supports these response frames:

```
static const uint8_t JBD_CMD_HWINFO = 0x03;
static const uint8_t JBD_CMD_CELLINFO = 0x04;
```

syssi, 2024-07-16:
I've pushed a small change at a feature branch. Please update your YAML from

```
substitutions:
  external_components_source: github://syssi/esphome-jbd-bms@main
```

to

```
substitutions:
  external_components_source: github://syssi/esphome-jbd-bms@liontron-support
``` 

to apply this change to your ESP. This is the difference: https://github.com/syssi/esphome-jbd-bms/pull/83/files

We are trying to avoid sending an probably unsupported command.

jassi0001, 2024-07-16:
Hi Sebastian, problem stays the same. It connects, reads values one time an than goes in BMS offline while the BLE Connection stays connected.

syssi, 2024-07-16:
Okay. As next step a btsnoop capture is required to get an idea whats different.

jassi0001, 2024-07-16:
will try to get an android mobile working....

jassi0001, 2024-07-21:
i'm sorry, i do not have an android mobile here available to do a btsnoop.... do you know a possibility with a raspberry pi? Or a Website, which discribes, what to do?

syssi, 2024-07-21:
No. I don't know another way to sniff the communication between the official app and the BMS.

syssi, 2024-07-25:
Could you provide a complete ESPHome log? I best case it will provide some more important details.

jassi0001, 2024-07-26:
I changed back to the "normal" Branch... 
[logs_testboard2_logs.txt](https://github.com/user-attachments/files/16393783/logs_testboard2_logs.txt)

syssi, 2024-07-26:
Could you provide another log using the feature branch? I would like to check the difference.

syssi, 2024-07-26:
I've extracted the important parts of the `main` branch log:

```
[17:32:28][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] Found device
[17:32:28][D][esp32_ble_tracker:669]: Found device 70:3E:97:D1:BE:20 RSSI=-92
[17:32:28][D][esp32_ble_tracker:690]:   Address Type: PUBLIC
[17:32:28][D][esp32_ble_tracker:692]:   Name: '0223400011'
[17:32:28][D][esp32_ble_tracker:695]:   TX Power: 2
[17:32:28][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[17:32:28][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[...]
[17:32:29][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] ESP_GATTC_SEARCH_CMPL_EVT
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x1800
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x1  end_handle: 0x7
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x1801
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x8  end_handle: 0xb
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x180A
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0xc  end_handle: 0xe
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0xFF00
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0xf  end_handle: 0x16
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 00010203-0405-0607-0809-0A0B0C0D1912
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x17  end_handle: 0x1b
[17:32:29][I][esp32_ble_client:227]: [0] [70:3E:97:D1:BE:20] Connected
[17:32:29][V][esp32_ble_client:069]: [0] [70:3E:97:D1:BE:20]  characteristic 0xFF01, handle 0x11, properties 0x12
[17:32:29][V][esp32_ble_client:069]: [0] [70:3E:97:D1:BE:20]  characteristic 0xFF02, handle 0x15, properties 0x6
[17:32:29][V][esp32_ble_client:121]: [0] [70:3E:97:D1:BE:20] gattc_event_handler: event=18 gattc_if=3
[17:32:29][D][esp32_ble_client:188]: [0] [70:3E:97:D1:BE:20] cfg_mtu status 0, mtu 128
[17:32:29][V][esp32_ble_client:121]: [0] [70:3E:97:D1:BE:20] gattc_event_handler: event=38 gattc_if=3
[17:32:29][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:32:29][D][esp32_ble_client:296]: Wrote notify descriptor 1, properties=18
[17:32:29][I][jbd_bms_ble:093]: Request device info
[17:32:29][VV][jbd_bms_ble:555]: Send command: DD.A5.05.00.FF.FB.77 (7)
(0x05 request ignored)

[17:32:31][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
[17:32:31][VV][jbd_bms_ble:103]: Notification received: DD.03.00.19.05.32.00.00.10.D7.11.30.00.01.2D.03.00.00.00.00.00.00.25.62.03.04.01.0B.95.FD.28.77 (12)
[...]
[17:32:31][I][jbd_bms_ble:260]: Hardware info frame (25 bytes) received
[17:32:31][VV][jbd_bms_ble:261]:   05.32.00.00.10.D7.11.30.00.01.2D.03.00.00.00.00.00.00.25.62.03.04.01.0B.95 (25)
[17:32:31][D][jbd_bms_ble:263]:   Device model: 
[17:32:31][D][sensor:094]: 'bms total voltage': Sending state 13.30000 V with 2 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms current': Sending state 0.00000 A with 1 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms power': Sending state 0.00000 W with 1 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms charging power': Sending state 0.00000 W with 2 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms capacity remaining': Sending state 43.11000 Ah with 2 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms nominal capacity': Sending state 44.00000 Ah with 2 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms charging cycles': Sending state 1.00000  with 0 decimals of accuracy
[17:32:31][D][jbd_bms_ble:290]:   Date of manufacture: 2022.8.3
[17:32:31][D][sensor:094]: 'bms balancer status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[17:32:31][D][binary_sensor:036]: 'bms balancing': Sending state OFF
[17:32:31][D][sensor:094]: 'bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[17:32:31][V][text_sensor:013]: 'bms errors': Received new state 
[17:32:31][D][text_sensor:064]: 'bms errors': Sending state ''
[17:32:31][D][sensor:094]: 'bms software version': Sending state 2.50000  with 1 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms state of charge': Sending state 98.00000 % with 0 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms operation status bitmask': Sending state 3.00000  with 0 decimals of accuracy
[17:32:31][D][binary_sensor:036]: 'bms charging': Sending state ON
[17:32:31][D][switch:055]: 'bms charging': Sending state ON
[17:32:31][D][binary_sensor:036]: 'bms discharging': Sending state ON
[17:32:31][D][switch:055]: 'bms discharging': Sending state ON
[17:32:31][D][sensor:094]: 'bms battery strings': Sending state 4.00000  with 0 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms temperature 1': Sending state 23.40000 °C with 1 decimals of accuracy

[17:32:31][VV][jbd_bms_ble:555]: Send command: DD.A5.04.00.FF.FC.77 (7)
[17:32:31][VV][jbd_bms_ble:103]: Notification received: DD.04.00.08.0C.FB.0C.FD.0C.FD.0C.FE.FB.D5.77 (15)
[17:32:31][I][jbd_bms_ble:209]: Cell info frame (8 bytes) received
[17:32:31][VV][jbd_bms_ble:210]:   0C.FB.0C.FD.0C.FD.0C.FE (8)
[17:32:31][D][sensor:094]: 'bms cell voltage 1': Sending state 3.32300 V with 3 decimals of accuracy
[17:32:31][D][sensor:094]: 'bms cell voltage 2': Sending state 3.32500 V with 3 decimals of accuracy
[... cut ...]
[17:33:16][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
[17:33:31][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
[17:33:46][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
[17:34:01][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
[17:34:16][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
[17:34:31][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
[17:34:46][VV][jbd_bms_ble:555]: Send command: DD.A5.03.00.FF.FD.77 (7)
```

syssi, 2024-07-26:
I've pushed some more changes to the feature branch (`liontron-support`). Please give it another try and provide the log. :-)

jassi0001, 2024-07-26:
it seems that its not copnnecting with the lion branch....
[logs_testboard2_logs (1).txt](https://github.com/user-attachments/files/16395615/logs_testboard2_logs.1.txt)

syssi, 2024-07-26:
It looks like you aren't using the most recent version of the `liontron-support` branch. 

See https://github.com/syssi/esphome-jbd-bms/pull/83/files

The `Send command (handle 0x%02X):` message should include the used handle.

jassi0001, 2024-07-26:
tried to clear the cache, but i dont think that the download was successfull.. . can you see in the logs which version i [downloaded?[](url)]([logs_testboard2_logs (2).txt](https://github.com/user-attachments/files/16396226/logs_testboard2_logs.2.txt))

syssi, 2024-07-26:
I've pushed another feature branch to trigger a clean build on your side:

```
substitutions:
  external_components_source: github://syssi/esphome-jbd-bms@liontron
```

Could you update your YAML and give it another try? Thanks in advance!

jassi0001, 2024-07-26:
looks very good....

syssi, 2024-07-26:
I've pushed some more changes. :-) Please recompile again and provide another log.

jassi0001, 2024-07-26:
seems to stay stable.... some invalid frame length... will take a look tomorrow and check, what homeassist is logging.....
[logs_testboard2_run.txt](https://github.com/user-attachments/files/16396780/logs_testboard2_run.txt)

Thank you very much for your support so far. I think it will help others, who have liontron as batterys... there was not much to find, before i found yout project....

syssi, 2024-07-26:
It's not perfect yet. Your BMS firmware has timing issues. I will try to make some more changes next days.

syssi, 2024-07-26:
About the timing: The Android app transmits the following request every 500ms:

```
# JBD_CMD_HWINFO
DD A5 03 00 FF FD 77
```

The second request is transmitted after `1105ms` the first time and every `3100ms` later on:

```
# JBD_CMD_CELLINFO
DD A5 04 00 FF FC 77
```

We will have to imitate this behavior/timing because the BMS firmware is unable to schedule responses properly.

syssi, 2024-07-26:
I've pushed another improvement. Does it still work?

jassi0001, 2024-07-27:
think no... is connected, but does not seem to get Values....

[logs_testboard2_logs.txt](https://github.com/user-attachments/files/16398921/logs_testboard2_logs.txt)

syssi, 2024-07-27:
Could you change the `update_interval` of the `jbd_bms_ble` to `1s`. Does the change something?

syssi, 2024-07-27:
You are using the `main` branch again. Please stick to `liontron` as long we are testing here.

jassi0001, 2024-07-27:
> You are using the `main` branch again. Please stick to `liontron` as long we are testing here.

dont think so.. since yesterday i did not change the source... it's still liontron.... 

  external_components_source: github://syssi/esphome-jbd-bms@liontron

jassi0001, 2024-07-27:
> Could you change the `update_interval` of the `jbd_bms_ble` to `1s`. Does the change something?

Update Interfall updates every second... Cell Voltage updates every update intervall... and stays connected.... it seems

syssi, 2024-07-27:
Sorry, my fault! I was looking on an outdated download/ESPHome log. So the connection keeps only up and running on a fast update interval? Could you play around with the value?

syssi, 2024-07-27:
Could you provide another ESPHome log using the `1s` update interval? I would like to the both frames (type 0x03 and 0x04) are received properly.

jassi0001, 2024-07-27:
Here the logs...

Just to let you know and dont confuse you... i have two Batteries, which i try....

 mac_address: 70:3e:97:D1:C1:8C - 55 Ah and two temp Sensor
 mac_address: 70:3E:97:D1:BE:20 - 40 Ah and one temp Sensor

Switching Input Charge and Output charge does not work as well... but thats next....
[logs_testboard2_logs (1).txt](https://github.com/user-attachments/files/16399076/logs_testboard2_logs.1.txt)

By the way: Don't you sleep at all? every time I write you reply within minutes... very fascinating...

jassi0001, 2024-07-28:
Hi Sebastian,

the last Version  you provided works for me with Update Interval of 8s. Since the Compile there was no Prooblem.

syssi, 2024-07-28:
That's a bit confusing because to difference to the `main` branch is pretty tiny. If you restart the ESP multiple times is the ESP always able to setup a proper BLE connection + data flow or does it stop every now and then?

jassi0001, 2024-07-28:
after a rebbot i have sometimes to reboot again.... but since i set to 8s there is no disconnect anymore.... i tried 10s, that did not work properly

syssi, 2024-07-28:
Yes. Let's try to main branch. This doesn't work out of the box, correct?

jassi0001, 2024-07-28:
strange.... it seems the only problem was se poll intervall ... with the main branchs it's starting and working as expected ..... dont understand that...

syssi, 2024-07-28:
You didn't tried the default in the past?

syssi, 2024-07-28:
I've merged some minor improvements of the Liontron branch nevertheless now into `main`. Let's declare this issue as resolved! :-)

jassi0001, 2024-07-28:
i initialy tried the main branch... everything starts working after changing the update intervall to lower than 15s... i just tried 9s which works as well.... last time i tried 10s it did not work...

so lets say its resolved.... hopefully the updateintervall chang3 helps some other guys.... Thans for your support, figuring this out.

syssi, 2024-07-28:
Last question: Could you provide some more details about your Liontron BMS? Do you know the exact model name? Could you tell me the name/model of your battery pack? I would like to add your battery / BMS to the list of supported devices.

syssi, 2024-07-28:
Notes for myself:

```
[17:32:28][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] Found device
[17:32:28][D][esp32_ble_tracker:669]: Found device 70:3E:97:D1:BE:20 RSSI=-92
[17:32:28][D][esp32_ble_tracker:690]:   Address Type: PUBLIC
[17:32:28][D][esp32_ble_tracker:692]:   Name: '0223400011'
[17:32:28][D][esp32_ble_tracker:695]:   TX Power: 2
[17:32:28][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[17:32:28][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[...]
[17:32:29][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] ESP_GATTC_SEARCH_CMPL_EVT
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x1800
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x1  end_handle: 0x7
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x1801
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x8  end_handle: 0xb
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x180A
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0xc  end_handle: 0xe
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0xFF00
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0xf  end_handle: 0x16
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 00010203-0405-0607-0809-0A0B0C0D1912
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x17  end_handle: 0x1b
[17:32:29][I][esp32_ble_client:227]: [0] [70:3E:97:D1:BE:20] Connected
[17:32:29][V][esp32_ble_client:069]: [0] [70:3E:97:D1:BE:20]  characteristic 0xFF01, handle 0x11, properties 0x12
[17:32:29][V][esp32_ble_client:069]: [0] [70:3E:97:D1:BE:20]  characteristic 0xFF02, handle 0x15, properties 0x6
[17:32:29][V][esp32_ble_client:121]: [0] [70:3E:97:D1:BE:20] gattc_event_handler: event=18 gattc_if=3
[17:32:29][D][esp32_ble_client:188]: [0] [70:3E:97:D1:BE:20] cfg_mtu status 0, mtu 128
[17:32:29][V][esp32_ble_client:121]: [0] [70:3E:97:D1:BE:20] gattc_event_handler: event=38 gattc_if=3
[17:32:29][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:32:29][D][esp32_ble_client:296]: Wrote notify descriptor 1, properties=18
```

jassi0001, 2024-07-28:
these are the Batterys.

They have the passibility to set a PIN i Think.... but i dont know how....
https://liontron.com/download/german/LISMART1255LX.pdf
https://liontron.com/download/german/LISMART1240LX.pdf

syssi, 2024-07-29:
I've implemented a `retrieve hardware version` button entity yesterday:

```
button:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    retrieve_hardware_version:
      name: "${name} retrieve hardware version"
```

Could you add the section to your YAML and press the button a few times (once per second). Does your BMS repond or is the `0x05` request always ignored?

jassi0001, 2024-07-29:
here ist the Log: Seems to fetch the Name / Model
[logs_testboard2_logs.txt](https://github.com/user-attachments/files/16418435/logs_testboard2_logs.txt)

syssi, 2024-07-29:
```
[21:55:29][D][jbd_bms_ble:245]:   Device model: 0923550208
[21:55:29][D][jbd_bms_ble:272]:   Date of manufacture: 2024.2.3
```

jassi0001, 2024-07-30:
Hi Sebastian, just to clarify. this is the Name of the Unit not the model Name. This name can be changed, I think.
A Model Name can not be read from the BMS…. the LIONTRON App does not read the Model as well. The he name you read is the name  the device is showing in BLE Scan…..

syssi, 2024-07-30:
Thanks for the clarification. Did you try to get the Mosfet switches working?

jassi0001, 2024-07-30:
The Switch worked as expected

syssi, 2024-07-30:
Perfect! Feel free to create new issues if you miss something or if you see room for improvements.

closed 2024-07-30 by syssi

jassi0001, 2025-01-06:
Hi Syssi,

we got the liontron System up and running. Another Project directly integrated into HA is trying the same and there is the same timing Problem as we hat in ESPHome. Can you give more input, what you have done to fix the Problem? The crosslink to the git artikel is this: https://github.com/patman15/BMS_BLE-HA/issues/141
Hope you can give a hint....

syssi, 2025-01-09:
I talked to Patrick and your issue is unrelated.

jassi0001, 2025-01-09:
thanks

Mit freundlichen Grüßen

Stefan Jasper
Breite Str. 4a

38159 Vechelde

015678-514093