syssi/esphome-jbd-bms #95: Add decimal places to SOC?

fboundy, 2024-08-29:
Hi - great work. I'm running this along with your Votronic repo and it's working very well.

Any way I can increase the decimal places on the state_of_charge sensor? Happy to edit in my fork if you want to keep it at zero in the main repo. I've edited it in sensors.py but now it's just reporting .0% all the time so I must be missing something.

syssi, 2024-08-29:
The BMS firmware exposes the `state of charge` as `uint8_t`. In other words: There are no decimal places available in the payload of the BMS.

fboundy, 2024-08-29:
Oh that's curious as I was previously using Batmon which showed 2 DPs as (I think) the Fogstar (battery brand) app did. Obviously if it's not in the BLE Characteristic

syssi, 2024-08-29:
It's `uint8_t` here for sure:

https://github.com/syssi/esphome-jbd-bms/blob/main/components/jbd_bms_ble/jbd_bms_ble.cpp#L287-L288

syssi, 2024-08-29:
Is the Fogstar app an android application? Could you provide a download link?

fboundy, 2024-08-29:
I was mistaken - the app only shows an integer. Batmon shows 2 DP but maybe it calculates it from Capacity Remaining and Capacity. Which I can always do with a template

syssi, 2024-08-29:
Perfect! I will close this issue. Feel free to create new issues if you miss something or if you see room for improvements.

closed 2024-08-29 by syssi

syssi, 2024-08-30:
In general you can override/change the `accuracy_decimals` of any sensor using the YAML configuration without touching the python files of the custom component:

```
sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    current:
      name: "${name} current"
      accuracy_decimals: 0
```