syssi/esphome-jbd-bms #97: battery cycle capacity not showing on SP22S003BL8S100A

RudolfRendier, 2024-09-12:
Hey,

Thanks for this software. It works really well.
I have loaded the example esp32-ble-example.yaml and I also have the below sensor configured. But it is nowhere to be found. The sensor isn't there at all.
```
    battery_cycle_capacity:
      name: "${name} battery cycle capacity"
 ```

Basen Green 24V 100 Ah SP22S003BL8S100A v7.6
Date of manufacture: 2024.5.24

Not that I really need it, because it's a fairly static 'value/setting'. But it surprised me it was missing.
Happy to provide more info if needed.

Maybe it's just missing from this list, but it's also some kind of 'setting', not really a sensor, so that may be on purpose.
https://github.com/syssi/esphome-jbd-bms/blame/8093fa138c52f3c9df4f0f2f34bf46fb5e08beaa/components/jbd_bms/sensor.py#L153C12-L153C12

syssi, 2024-09-12:
This sensor was introduced accidently and isn't available as data point at the JBD protocol. The static setting you are looking for is probably the `nominal_capacity`.

RudolfRendier, 2024-09-13:
Thanks for the follow-up.
I do have nomimal_capacity; and it's at 100Ah.
Strange is that the Overkill Solar app also reports a Cycle Capacity of 80Ah, next to the nominal capacity of 100Ah.

According to the [manual ](https://raw.githubusercontent.com/FurTrader/OverkillSolarBMS/master/Overkill_Solar_BMS_Instruction_Manual.pdf)it can be set to the actual, real-world, capacity of the battery.

> 3.2 Capacity Parameters
These parameters are used to display the battery capacity and to calculate the state of charge..
3.2.1 Designed Capacity
This should be set to the battery pack’s capacity, in amp hours (Ah). It is not used to calculate the state of
charge. It’s simply used when displaying the intended capacity of the cells. The actual capacity of the pack is
defined as the cycle capacity, which is described in the next section. Designed capacity can be calculated as
follows:
Designed capacity=Cell capacity Parallel cell count of pack
NOTE: This parameter is displayed in milliamp hours in the iPhone app.
3.2.2 Cycle Capacity
This parameter is used to calculate state of charge. In the real world, batteries do not meet the designed
capacity printed on the cells. It can be higher, if the cell was underrated, or it can be lower (especially true for
used or B- or C-grade cells).
Ideally, the capacity of the battery pack should be measured, and the actual number should be programmed
into the BMS. Therefore:
Cycle capacity =Actual measured capacity of the pack
There are several ways to measure the pack’s total capacity. The easiest way is to set the cycle capacity to
hook a known DC load up to the battery, and measure the amount of time it takes from full charge down to
cutoff. Ensure that the protection parameters are set before starting. Charge up to 100% (see the section on
top-balancing). Note the start time. Record the pack voltage in 15 minute intervals (this info may be used to
set the percent capacity voltages, later in this section). Record the time when the BMS protection circuitry
cuts off the discharge current.
Cycle capacity = Test load current (A) Total run time (h)
NOTE: This parameter only affects the state of charge. If it is set too low, the state of charge will hit zero
percent before the battery is actually at zero percent. The state of charge percentage will never go negative

I'll try to see what happens if I change the value; i.e. if it persists across restarts of the app.
And if it impacts the SoC reported in the app or via BLE.

syssi, 2024-09-13:
Good point! Which Android app do you use? I would like to get an idea it's an firmware feature of the BMS or something virtually provided by the App.