syssi/esphome-jbd-bms #100: Build HA entity_id from id of the object

ThomasCr, 2024-10-01:
Hi,

today I started with this nice project and wanted to translate and rename some items for me. What I don't like, when the `entity_id' changes (this happens if I change the name attribute). So I defined for every "sensor/button/..." a own id. But at the end, I noticed, that those IDs was not used in HA.

(small example where I am)
```
sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    battery_strings:
      id: "jbd_bms0_battery_strings"
      name: "${bms0_name} Batteriestränge"
    current:
      id: "jbd_bms0_current"
      name: "${bms0_name} Stromstärke"
    power:
      id: "jbd_bms0_power"
      name: "${bms0_name} Leistung"
```

Can you generate the id in your code? Or allow us to define this attribute?

When I try to define it, I get the message: 

> [entity_id] is an invalid option for [battery_strings]. Did you mean [mqtt_id]?

In your code are also all the other HA attributes defined.

So it should be not such a big problem to do (pseudo code):
```
if id defined:
  set  entity_id = sensor.$id
```

syssi, 2024-10-01:
It looks like this is your first contact with ESPHome. The project is a custom component to talk to a JBD BMS. The behavior of sensor entities is defined at the ESPHome core. Please use Discord to talk to the community and discuss your ideas here. This project is only implements the JBD protocol and uses ESPHome as underlying framework.

syssi, 2024-10-01:
Long story short: You have to use the `name` attribute to define the entity_id. The speaking name must be applied in Home Assistant manually or vice versa. AFAIK you cannot set both using the YAML configuration at the moment.

syssi, 2024-10-01:
The `id` is used to reference entities internally (at ESPHome) but isn't passed to Home Assistant.

ThomasCr, 2024-10-01:
But in your code, you also build the HA attributes:

https://github.com/syssi/esphome-jbd-bms/blob/585be3b1751af17b63e3abc1473f413052a127b3/components/jbd_bms_ble/sensor.py#L181

are your sure, that it is not possible from your custom component to build / define the entity_id for HA?

syssi, 2024-10-01:
You can find the different attributes of the `sensor_schema` here: https://github.com/esphome/esphome/blob/dev/esphome/components/sensor/__init__.py#L308-L335

The `SENSOR_SCHEMA` is defined here: https://github.com/esphome/esphome/blob/dev/esphome/components/sensor/__init__.py#L260-L303

And the `ENTITY_BASE_SCHEMA` is defined here: https://github.com/esphome/esphome/blob/dev/esphome/config_validation.py#L1912-L1928

AFAIK you can pass just a entity name here which is used (in snake case) as `entity_id` and as the speaking name default of the sensor.

ThomasCr, 2024-10-01:
to bad, I also found the docs of the sensor component:
https://esphome.io/components/sensor/index.html

But this custom compontent can define the entity_id (not tested, just found it right now)
https://community.home-assistant.io/t/custom-component-cover-time-based/187654

syssi, 2024-10-02:
Let's close the issue because this cannot be improved here for sure.

closed 2024-10-02 by syssi