syssi/esphome-jbd-bms #101: SP16S020L16S100A: Missing attributes

ThomasCr, 2024-10-01:
Hi,

I have a new battery with a BT BMS:

It is a [Vatrer Power 51.2v 100ah LiFePo4](https://youtu.be/1n8ZkgtLBt0?si=UM3RFA2l9P-JnBKO&t=627) battery - the BMS should be [Jiabaida / JBD SP16S020](https://jiabaida-bms.com/products/jbd-smart-bms-8s-16s-6-10s-60a-80a-100a-lifepo4-li-ion-for-lithium-iron-battery-pcb-with-uart-rs485-function-bt-module-with-balance-wires)

Informations from Overkill Solar App:
Manufacturer: ZJDY
Firmware Version: 6.2
Device Name: SP16S020L16S100A

The SP16S020 Modell should be tested (its listed in the Readme) - but not with the same Firmware

Some of the attributes are in state "unknown":
- text_sensor.device_model
- text_sensor.operation_status
- binary_sensor.discharging

Also a small questions stays: why is there a Button for retrieve_hardware_version? Why not fetch this info like the other informations and put them in a text_sensor field?

syssi, 2024-10-01:
There are BMS models which doesn't support the `model version` request and behaves pretty strange afterwards. The value is pretty useless in general so I try to avoid the retrieval.

syssi, 2024-10-01:
How do you talk to your BMS? Using UART or BLE? Could you increase the log level and provide some logs including the RAW traffic. I would like to have a look why the mentioned measurements/states are missing.

ThomasCr, 2024-10-01:
I connect with BLE - I set the debug params from `esp32-ble-example-debug.yaml` - hope thats good.

[jbd_bms_ble.log](https://github.com/user-attachments/files/17215804/jbd_bms_ble.log)

ThomasCr, 2024-10-03:
I also noticed, that I cant switch off the switch.discharging switches automatically back, when I switch if manually off in HomeAssistant.

closed 2024-12-21 by syssi