syssi/esphome-jbd-bms #105: rs485 to d1 mini

IvanSerevko, 2024-10-23:
Hello, is it possible to connect this not via UART, but via RS485? I tried to configure it for different pins and do not receive data (I connected both GPIO4,GPIO5 and via RX,TX, I have the following model JBD-AP21S001 200A)

[image]

syssi, 2024-10-23:
It depends on your hardware. What kind of signal is available at the RX/TX pins of your BMS?

IvanSerevko, 2024-10-23:
[image]
also tried through
[image]
and shows the data on the computer
[image]
and near the rs485 pin there are also 5v and gnd from which the wemos d1 mini was powered

syssi, 2024-10-23:
Could you provide a macro/high resolution photo of the USB-to-RS485 converter module? If we are able to identify a RS485 converter here we are sure a RS485 converter is required between the ESP and the BMS.

IvanSerevko, 2024-10-23:
[image]

syssi, 2024-10-23:
Okay, this is a RS485 signal for sure. You have to attach a RS485 converter module to your ESP to interface the BMS. You should use one with automatic flow control like this:

[image]

IvanSerevko, 2024-10-23:
Thank you very much, I will try it and let you know if it worked

IvanSerevko, 2024-10-29:
everything works great, thanks a lot, next task will be how to combine it with Solar2Mqqt)

closed 2024-10-29 by IvanSerevko

syssi, 2024-10-29:
Which BMS model do you own? I would like to extend the list of supported devices if possible.