syssi/esphome-jbd-bms #110: Webserver v3 grouping and sorting support

universam1, 2024-12-11:
Thank you for this amazing project!

Following issue I can see with the https://esphome.io/components/web_server.html#version-3-features grouping and sorting, that it is not supported and raising an error:

```
sensor.jbd_bms_ble: [source .\mpptbms.yaml:129]
  platform: jbd_bms_ble
  jbd_bms_ble_id: bms0

  [web_server] is an invalid option for [sensor.jbd_bms_ble]. Please check the indentation.
  web_server:
    sorting_group_id: sorting_group_bms_settings
```

syssi, 2024-12-12:
Could you share your YAML configuration? It looks like you have introduced the issue while customizing the configuration.

universam1, 2024-12-12:
@syssi thanks for checking - please find attached the file

<details>

<summary>yaml file</summary>

```yaml
substitutions:
  name: BMS
  mac_address_bms: A4:C1:38:E6:97:0C
  mac_address_mppt: fcccb997060c
  bindkey_mppt: 9a131717779c5460002e224654515e3d

esphome:
  name: mobibluectrl
  friendly_name: MobiBlue Controller
  comment: Controlling JBD Battery Management System and Victron SmartSolar MPPT via BLE

external_components:
  - source: github://Fabian-Schmidt/esphome-victron_ble
  - source: github://syssi/esphome-jbd-bms@main

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
logger:
  level: WARN
  logs:
    jbd_bms_ble: ERROR
wifi:
  ssid: asdf
  password: asdf

  # ap:
  #   ap_timeout: 1s
  # reboot_timeout: 0s
web_server:
  port: 80
  version: 3
  sorting_groups:
    - id: sorting_group_mppt_settings
      name: "Victron MPPT Solar"
      sorting_weight: 10
    - id: sorting_group_bms_settings
      name: "BMS"
      sorting_weight: 20

ota:
  platform: esphome

esp32_ble_tracker:
  scan_parameters:
    active: false

victron_ble:
  - id: SmartSolar
    mac_address: ${mac_address_mppt}
    bindkey: ${bindkey_mppt}

ble_client:
  - id: client0
    mac_address: ${mac_address_bms}

jbd_bms_ble:
  - id: bms0
    ble_client_id: client0
    update_interval: 1s

binary_sensor:
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT is in Fault state"
    type: DEVICE_STATE_FAULT
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT has Error"
    type: CHARGER_ERROR

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    balancing:
      name: "${name} balancing"
    # charging:
    #   name: "${name} charging"
    # discharging:
    #   name: "${name} discharging"
    online_status:
      name: "${name} online status"

sensor:
  # SmartSolar
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT Output Voltage"
    type: BATTERY_VOLTAGE
    web_server:
      sorting_group_id: sorting_group_mppt_settings
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT Output Current"
    type: BATTERY_CURRENT
    web_server:
      sorting_group_id: sorting_group_mppt_settings
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT Output Power"
    type: BATTERY_POWER
    web_server:
      sorting_group_id: sorting_group_mppt_settings
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT Yield Today"
    type: YIELD_TODAY
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT PV Power"
    type: PV_POWER
    web_server:
      sorting_group_id: sorting_group_mppt_settings
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT Load Current"
    type: LOAD_CURRENT
    web_server:
      sorting_group_id: sorting_group_mppt_settings

  - platform: jbd_bms_ble
    web_server:
      sorting_group_id: sorting_group_bms_settings
    jbd_bms_ble_id: bms0
    # battery_strings:
    #   name: "${name} cell count"
    current:
      name: "${name} current"
    # power:
    #   name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    state_of_charge:
      name: "${name} state of charge"
    # nominal_capacity:
    #   name: "${name} nominal capacity"
    # charging_cycles:
    #   name: "${name} charging cycles"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_voltage:
      name: "${name} total voltage"
    # average_cell_voltage:
    #   name: "${name} average cell voltage"
    delta_cell_voltage:
      name: "${name} cell delta voltage"
    # min_cell_voltage:
    #   name: "${name} min cell voltage"
    # max_cell_voltage:
    #   name: "${name} max cell voltage"
    # min_voltage_cell:
    #   name: "${name} min voltage cell"
    # max_voltage_cell:
    #   name: "${name} max voltage cell"
    temperature_1:
      name: "${name} temperature"
    temperature_2:
      name: "Battery temperature"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    # software_version:
    #   name: "${name} software version"

    # operation_status_bitmask:
    #   name: "${name} operation status bitmask"
    # errors_bitmask:
    #   name: "${name} errors bitmask"
    # balancer_status_bitmask:
    #   name: "${name} balancer status bitmask"

text_sensor:
  - platform: victron_ble
    victron_ble_id: SmartSolar
    name: "MPPT state"
    type: DEVICE_STATE
  # - platform: victron_ble
  #   victron_ble_id: SmartSolar
  #   name: "MPPT Error reason"
  #   type: CHARGER_ERROR

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    # errors:
    #   name: "${name} errors"
    operation_status:
      name: "${name} operation status"
    # device_model:
    #   name: "${name} device model"

  # - platform: ble_scanner
  #   name: "BLE Devices Scanner"

switch:
  - platform: ble_client
    ble_client_id: client0
    name: "BLE enable BMS connection"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

```

</details>

closed 2024-12-12 by syssi

universam1, 2024-12-13:
@syssi thank you for the hints!