syssi/esphome-jbd-bms #112: not able to connect to more then 3 BLE clients

JanStry, 2025-01-01:
Hi, i have a problem when i connect more then 3 clients at the same time.
when i manually disconnect one or more working clients the 4th client connects.

here are some logs:

[00:44:21][D][esp-idf:000][BTU_TASK]: W (71132) BT_HCI: btu_hcif_hdl_command_status,opcode:0x200d,status:0x09

[00:44:21][D][esp-idf:000][BTU_TASK]: W (71137) BT_APPL: gattc_conn_cb: if=3 st=0 id=771 rsn=0x100

[00:44:21][D][esp-idf:000][BTU_TASK]: W (71140) BT_APPL: gattc_conn_cb: if=4 st=0 id=772 rsn=0x100

[00:44:21][D][esp-idf:000][BTU_TASK]: W (71152) BT_APPL: gattc_conn_cb: if=5 st=0 id=773 rsn=0x100

[00:44:21][D][esp-idf:000][BTU_TASK]: W (71161) BT_APPL: gattc_conn_cb: if=6 st=0 id=774 rsn=0x100

[00:44:21][W][esp32_ble_client:142]: [2] [A5:C2:37:2A:06:37] Connection failed, status=133

------------------------------------------------------------------------------------------------------------------

btu_hcif_hdl_command_status,opcode:0x200d,status:0x09:
 according to espidf 0x09  error means :" BLE Connection Limit Exceeded"

however i am sure esphome ESP32 can connect to at least 7 BLE devices at the same time...
maybe there is some limiter set in the BLE library?

kind regards

syssi, 2025-01-01:
Which ESP32 model do you use? Did you try to re-order the different devices / ble_client connections just for testing?

In the past there was a hard limit at 3 simultaneous connections: https://github.com/esphome/esphome/commit/1a71cc304740f7823ca77eb3bbab7128f7dd612a

Do you use ESP-IDF as platform? Could you share your YAML configuration?

JanStry, 2025-01-02:
Hi, im using the ESP32doit devkit v1.
 this is my YAML config:

substitutions:
  name: xxx
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  bms2: "${name} bms2"
  bms3: "${name} bms3"
  device_description: "Monitor and control a Xiaoxiang Battery Management System (JBD-BMS) via BLE"
  external_components_source: github://syssi/esphome-jbd-bms@main
  bms0_mac_address: A5:C2:37:36:D9:xx
  bms1_mac_address: A5:C2:37:36:CA:xx
  bms2_mac_address: A5:C2:37:2A:06:xx
  bms3_mac_address: A5:C2:37:2A:0A:xx
  
esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-jbd-bms"
    version: 2.0.0

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: "xxxxxxxxx"
  password: "xxxxxxxx"

ota:
  platform: esphome
  on_begin:
    then:
      - switch.turn_off: ble_client_switch1
      - switch.turn_off: ble_client_switch2
      - switch.turn_off: ble_client_switch3
      - switch.turn_off: ble_client_switch0      
      - logger.log: "BLE connection suspended for OTA update" #added this for smoother OTA uploads

logger:
  level: DEBUG
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

api:

esp32_ble_tracker:
  scan_parameters:
    active: false

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
[... 183 lines cut ...]
    charging:
      name: "${bms0} charging"
    discharging:
      name: "${bms0} discharging"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms1
    charging:
      name: "${bms1} charging"
    discharging:
      name: "${bms1} discharging"
      
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms2
    charging:
      name: "${bms2} charging"
    discharging:
      name: "${bms2} discharging"
      
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms3
    charging:
      name: "${bms3} charging"
    discharging:
      name: "${bms3} discharging"

syssi, 2025-06-27:
If you upgrade your ESP to an ESP32S3 there should be some additional connections possible.

closed 2025-06-27 by syssi