syssi/esphome-jbd-bms #113: time to go

Daswomo, 2025-01-09:
you can do it?

time to go

syssi, 2025-01-10:
I don't understand what you are looking for.

DanClarke-io, 2025-01-29:
I think they are asking for a estimated time until the battery is depleted, for example in the Fogstar app they give you:

a remaining time:
[image]

and a charge time:
[image]

I assume this may be calculated from the Nominal current capacity and current current draw?

DanClarke-io, 2025-01-29:
Looking at this I think its mostly what #109 was, and as mentioned there, should be calculated as a template sensor within HA

syssi, 2025-01-29:
ESPHome supports template sensors too. Home Assistant isnt a requirement here.

syssi, 2025-07-22:
I'm adding to new template sensors to the project. It will looks like this:

[image]

Please give it a try! @DanClarke-io @Daswomo @Benni1123

closed 2025-07-23 by syssi