syssi/esphome-jbd-bms #114: Buttons for Switch On/Off (dis)charge Swaps back, after trigger it

ThomasCr, 2025-01-14:
Hi,

I like to use the buttons for switch on/off the charge/discharge of the BMS. My problem is, that sometimes (mostly) they switch back.

https://github.com/user-attachments/assets/c8a17e3d-258c-4650-804d-5756bfd24767

Dont know, whats wrong....

syssi, 2025-01-15:
Do you use a BLE or UART connection? Could you increase the log level, repeat the experiment / reproduce the issue a few times and provide the log? It looks like your BMS doesnt accept the instruction.

stalniy, 2025-02-10:
In my case, BMS blocks disabling charging when it has additional load (except of inverter) but once inverter is on grid power, BMS allows to switch off charging