syssi/esphome-jbd-bms #119: Add sensors for error counts

nagi121, 2025-02-04:
The EEPROM Register provides read only values for the various error counts. 
It would be nice to have these error counts as sensors in HA.

<table>
<tr>
<th>Register Address</th>
<th>Register Name</th>
<th>Data</th>
<th>Format</th>
<th>Unit</th>
<th>Field name(s)</th>
<th>Notes</th>
</tr>
<tr>
<td>0xAA</td>
<td>error_cnts</td>
<td>Various error condition counts</td>
<td>11 U16</td>
<td>

\--
</td>
<td>sc_err_cnt, chgoc_err_cnt, dsgoc_err_cnt, covp_err_cnt, cuvp_err_cnt, chgot_err_cnt, chgut_err_cnt, dsgot_err_cnt, dsgut_err_cnt, povp_err_cnt, puvp_err_cnt</td>
<td>READ ONLY</td>
</tr>
</table>

syssi, 2025-02-04:
```
sc_err_cnt: Short Circuit Error Count
chgoc_err_cnt: Charge Over Current Error Count
dsgoc_err_cnt: Discharge Over Current Error Count
covp_err_cnt: Cell Over Voltage Protection Error Count
cuvp_err_cnt: Cell Under Voltage Protection Error Count
chgot_err_cnt: Charge Over Temperature Error Count
chgut_err_cnt: Charge Under Temperature Error Count
dsgot_err_cnt: Discharge Over Temperature Error Count
dsgut_err_cnt: Discharge Under Temperature Error Count
povp_err_cnt: Pack Over Voltage Protection Error Count
puvp_err_cnt: Pack Under Voltage Protection Error Count
```

Does these descriptions of the sensors you are looking for look good?

nagi121, 2025-02-04:
I found two other descriptions:

[image]

[image]

I guess, your descriptions are much better. Maybe the word "Pack" should be better "Batterie"?

syssi, 2025-02-04:
I made some progress here (#120). There will be a `select` entity to retrieve selected eeprom register in future. Only the new sensor entities are still missing. In best case I will finish the feature later the week.

closed 2025-02-05 by syssi

nagi121, 2025-02-05:
Works perfectly. Great job!