syssi/esphome-jbd-bms #121: JBD RS485 with the LilyGo T-CAN485

andrewfraley, 2025-02-06:
I thought I'd drop a note here that I was able to get this working with a LilyGo T-CAN485 connected to a JBD-AP21S002 via the RS485 pins on the BMS.  I didn't have to change anything other than:

```
  tx_pin: GPIO22
  rx_pin: GPIO21
```

Thanks!

syssi, 2025-02-06:
To enable the RS485 converter of the T-CAN485 you have to pull up some GPIOs normally:

```
output:
  # Enable the RS485 converter
  - platform: gpio
    id: ENABLE_PIN
    pin:
      number: GPIO19
      inverted: true

  # Enable autodirection
  - platform: gpio
    id: SE_PIN
    pin:
      number: GPIO17
      inverted: true

  # Enable 5V pin of the RS485 converter
  - platform: gpio
    id: ENABLE_5V_PIN
    pin:
      number: GPIO16
      inverted: true
```

Are you sure it's a RS485 connection at the moment and not UARTL-TTL only?

andrewfraley, 2025-02-06:
Ah so I did a hard reboot / power cycle and it stopped working, I had turned those outputs on messing around with a different approach.  You do in fact need to add your above example.  Thanks!

syssi, 2025-02-06:
Alright! In this case we can be sure it's a RS485 connection. Thanks for your feedback!