syssi/esphome-jbd-bms #126: can't establish a bluetooth connection to 1 of 3 batteries

knogochill, 2025-03-19:
Hello,

I initially configured my ESP32 device and everything was running. It was connected to all 3 batteries. After adding some sensors to the configuration and reflashing the ESP32 only 2 connections are working and 1 not. But I can connect via the XiaoXiang app to the missing batterie. I already deleted the device in Home Assistant and created it again, used another ESP32 device but nothing works. 

[W][jbd_bms_ble:170]: [A4:C1:37:24:XX:XX] Not connected

syssi, 2025-03-19:
Do you use a ESP32 per BMS or do you try to talk to three devices using a single ESP?

knogochill, 2025-03-19:
I use 1 ESP for all devices. 

Also when I only try to connect to the missing one and delete the code of the 2 working ones it does not connect.

syssi, 2025-03-19:
Could you provide a log (while talking to the problematic BMS only)? I would like too possible error messages, BLE disconnect reasons, connection attempts, signal strength, etc.

knogochill, 2025-03-20:
Hello,
I only had one BMS configured, which didn't work at first. I let it run and after about 2 hours it suddenly worked. Then I added the other 2 BMS to the configuration and as if by magic all 3 BMS are working again. No idea where exactly the error was. Thanks for the help though. ❤️