syssi/esphome-jbd-bms #127: Cannot see data if amp is between -1 and 1.

DanClarke-io, 2025-04-07:
At the moment, it seems if there is less than 1amp being used (or generated) I get no output, so 1.1 amp will then say 13V or something along those lines, but if its less than 1 amp it says 0 for both. Does that make sense?

So normally fine:
[image]

But, if output is between -1amp and 1amp, the shunt continues to report it but the battery says 0:
[image]

And it's fine if its below -1amp too:
[image]

I may have set this up wrong, but thought I should see if I am alone!

syssi, 2025-04-08:
Could you show your YAML configuration? The `current` and `power` sensor will be sufficient. Is any filter applied here?

The component itself doesn't do any computation on the raw value: https://github.com/syssi/esphome-jbd-bms/blob/main/components/jbd_bms_ble/jbd_bms_ble.cpp#L270-L273

I guess it's a firmware/hardware limitation and cannot be resolved.

DanClarke-io, 2025-04-08:
Hi @syssi,

Current and power are pretty standard:
```
    current:
      name: "Current"
    power:
      name: "Power"
```

Whole YAML is below, I think I agree with you, must be a limitation of the device (this is a 300A Fogstar Drift Pro).
```
substitutions:
  name: "esp-battery-comms"
  device_description: "Monitor and control a Xiaoxiang Battery Management System (JBD-BMS) via BLE"
  external_components_source: github://syssi/esphome-jbd-bms@main

esphome:
  name: "esp-battery-comms"
  friendly_name: JDB Battery Comms

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on

logger:
  level: DEBUG

api:
  encryption: 
    key: REDACTED

ota:
  - platform: esphome
    password: !secret ota_password

esp32_ble_tracker:
  scan_parameters:
    active: false

ble_client:
  - id: client0
    mac_address: ${mac_address}

jbd_bms_ble:
  - id: bms0
    ble_client_id: client0
    update_interval: 2s

button:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    retrieve_hardware_version:
      name: "Retrieve Hardware Version"
  - platform: restart
    name: "Restart ESP"

binary_sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
[... 168 lines cut ...]
    errors:
      name: "Errors"
    operation_status:
      name: "Operation Status"
    device_model:
      name: "Device Model"

switch:
  - platform: ble_client
    ble_client_id: client0
    name: "Enable Bluetooth Connection"

  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    charging:
      name: "Charging"
    discharging:
      name: "Discharging"

```
 

Thanks,

Dan

syssi, 2025-04-09:
Alright! Let's close this issue because it cannot be resolved here.