syssi/esphome-jbd-bms #132: Deprecated text_sensor.TEXT_SENSOR_SCHEMA, switch.SWITCH_SCHEMA, button.BUTTON_SCHEMA

jdscavo, 2025-05-24:
Deprecation warnings are showing when using ESPHome 2025.5.0. Seems to be similar issue as noted in @syssi 
 https://github.com/syssi/esphome-jk-bms open issue 736 and closed issue 738 after updating to ESPHome 2025.5.0. Many of the deprecation warnings are repeated.  I think this is a list of the unique warnings in my use. number.NUMBER_SCHEMA warnings are not seen in the logs as reported in the other project:

```
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/bms-master.yaml...
INFO Updating https://github.com/syssi/esphome-jbd-bms.git@main
WARNING Using `button.BUTTON_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `button.button_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: jbd_bms_ble
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. 
WARNING Using `switch.SWITCH_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `switch.switch_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it.
```
In issue 736 of project syssi/esphome-jk-bms @syssi comments: There are also users reporting the entity names has changed which also causes confusion. 
I think this comment is a separate issue and not caused by the deprecation warnings above. I have an issue with entity names changing before ESPHome 2025.5.0 was released. The new deprecation warnings don't seem to correlate with what seems to be a totally different issue regarding entity names changing. I would be glad to post more on the name issue if interested.

BTW: Awesome work on these projects.

closed 2025-05-25 by syssi

syssi, 2025-05-25:
I've fixed the deprecation warning. Please give it a try!

jdscavo, 2025-05-27:
Your are awesome! Works great. 
Thank you!