syssi/esphome-jbd-bms #138: calculate state of charge manually or better alternative?

b2un0, 2025-06-11:
hi there,

first of all thanks for your work. using your component now for over a year.

currently im struggling with an issue, that the "state_of_charge" sensor remains on 99%.

i have a Model DP04S007L4S100A with 100A 4 strings.

[image]

[image]

[image]

[image]

[image]

i have some low power consumers, like the esp32 self, a "Truma CP Plus" for the Heater and an LTE Router.

if i calculate all 4 cells 
Cell 1: 3.316 V
Cell 2: 3.315 V
Cell 3: 3.317 V
Cell 4: 3.315 V

i got 13.263 V (also not matches the "total voltage" sensor)

for now, im using a template sensor like this

```yaml
sensor:
  - platform: jbd_bms_ble
    jbd_bms_ble_id: bms0
    battery_strings:
      name: "JBD BMS battery strings"
      id: jdb_bms_battery strings
    current:
      name: "JBD BMS current"
      id: jdb_bms_current
    power:
      name: "JBD BMS power"
      id: jdb_bms_power
    charging_power:
      name: "JBD BMS charging power"
      id: jdb_bms_charging_power
    discharging_power:
      name: "JBD BMS discharging power"
      id: jdb_bms_discharging_power
    state_of_charge:
      name: "JBD BMS state of charge"
      id: jdb_bms_soc
    nominal_capacity:
      name: "JBD BMS nominal capacity"
      id: jdb_bms_nominal_capacity
  - platform: template
    name: "JBD BMS State of Charge (Voltage)"
    id: vince_jbd_bms_soc_voltage
    unit_of_measurement: "%"
    state_class: "measurement"
    device_class: "battery"
    update_interval: 10s
    lambda: |-
      float voltage = id(bms_total_voltage).state;
      if (voltage >= 14.4) { return 100; }
      else if (voltage >= 14.0) { return 99; }
      else if (voltage >= 13.8) { return 98; }
      else if (voltage >= 13.6) { return 97; }
      else if (voltage >= 13.55) { return 96; }
      else if (voltage >= 13.5) { return 95; }
      else if (voltage >= 13.45) { return 90; }
      else if (voltage >= 13.4) { return 85; }
      else if (voltage >= 13.35) { return 80; }
      else if (voltage >= 13.3) { return 75; }
      else if (voltage >= 13.25) { return 70; }
      else if (voltage >= 13.2) { return 65; }
      else if (voltage >= 13.15) { return 60; }
      else if (voltage >= 13.1) { return 55; }
      else if (voltage >= 13.05) { return 50; }
      else if (voltage >= 13.0) { return 45; }
      else if (voltage >= 12.9) { return 40; }
      else if (voltage >= 12.8) { return 30; }
      else if (voltage >= 12.7) { return 25; }
      else if (voltage >= 12.6) { return 20; }
      else if (voltage >= 12.0) { return 15; } // Hier fällt die Spannung dann schnell
      else if (voltage >= 11.5) { return 10; }
      else if (voltage >= 11.0) { return 5; }
      else { return 0; }
```

are there better solutions for this?

Have I missed something?

b2un0, 2025-06-11:
im also looking into #63, could this be a solution?

b2un0, 2025-06-11:
hey, thanks.

with your latest changes 30minutes ago the button works!

i retrieved this in the logs
[09:17:40][W][jbd_bms_ble:199]: Unhandled response (function 0x0A) received: 

but the state of charge now shows the real value of 

[image]

syssi, 2025-06-11:
Nice! I will handle the response to complete the feature.