syssi/esphome-jbd-bms #149: does not show the correct data

br3ily, 2025-06-29:
Since the last update of esphome, it does not show the correct data in home assistant. It shows the data correctly in the HA graph and in the esphome logs.

[image]

[logs_jbd_logs.txt](https://github.com/user-attachments/files/20967336/logs_jbd_logs.txt)

syssi, 2025-06-29:
See 

https://github.com/home-assistant/core/issues/146873
https://github.com/esphome/issues/issues/7120
https://github.com/syssi/esphome-jk-bms/issues/755
https://github.com/syssi/esphome-tianpower-bms/issues/54

This is a workaround: https://github.com/syssi/esphome-jk-bms/issues/755#issuecomment-3016354434

syssi, 2025-07-03:
This is the pull request to resolve the issue: https://github.com/home-assistant/core/pull/147849/files

It's part of HA core 2025.7.

closed 2025-07-03 by syssi